open-rmf/rmf_traffic_editor #1: Map rescale only happens after editor restart

cnboonhan94, 2019-11-10:
Currently, when we first load an empty map, any annotations are extremely out of proportion with the current view. This scale issue is fixed on subsequent loads. It would be helpful to have this scaling done automatically.

codebot, 2019-11-12:
hopefully e3412c73cfa1e3bcbe86c7908a358be2adcc84fa and other recent commits resolves this; the default now is to start with a 10m x 10m view. Need to implement the `Level->Edit...` dialog box now in order to allow nice resizing later and/or adding an image. Can always edit the YAML by hand, but that's not particularly elegant.

codebot, 2019-12-17:
This should be fixed now. If not, feel free to re-open.

closed 2019-12-17 by codebot