open-rmf/rmf_traffic_editor #2: Import image feature is useful to create a template to trace over

DaikiMaekawa, 2019-11-11:
The users may already have an existing image that they want to trace over. In my case, it's an occupancy grid map which is in png format.

codebot, 2019-11-17:
#4 gets us some of the way towards having a reasonable way to import images using relative filenames. Up until now, we have been doing this by just manually entering the filename of the existing image(s) in the YAML file. It's not fully working yet in the GUI, but the pieces are starting to come together to make this process all GUI-driven now. Editing YAML by hand is fun and all, but when it's all GUI it will be super fancy :sparkles:

codebot, 2020-03-02:
I think this is now functional. Feel free to re-open if something doesn't work right.

closed 2020-03-02 by codebot