open-rmf/rmf_traffic_editor #6: migrate to a standard file format such as GeoJSON

codebot, 2019-11-20:
The current file format is just a random splat of structures to YAML. This was just done to get up and running quickly.

[Suggestion](https://discourse.ros.org/t/deterministic-navigation-in-ros/11442/12) by @ablakey to move to GeoJSON seems reasonable due to its use in many existing tools, such as QGIS. At the end of the day we just need to represent some spatial shapes and annotations in some file format.

That will require using JSON parsers in C++ (for the current "offline" Qt GUI) and Python (for the generators). I'm sure there are many options. This one looks nice for C++:  https://github.com/nlohmann/json  but the goal is to pick one that is already packaged up in Ubuntu 18, so we don't have to do it ourselves, or add another source dependency.

gavanderhoorn, 2019-11-20:
> I'm sure there are many options. This one looks nice for C++: https://github.com/nlohmann/json

Yes, it does, and it is.

> but the goal is to pick one that is already packaged up in Ubuntu 18

Seems it is: [nlohmann-json-dev](https://packages.ubuntu.com/search?keywords=nlohmann-json-dev), for Bionic, Cosmic and Disco.

codebot, 2019-11-20:
oh awesome, thanks @gavanderhoorn  !

codebot, 2021-10-04:
Experimenting now with `GeoPackage` http://geopackage.org  It seems pretty great so far.