open-rmf/rmf_traffic_editor #15: Feature request: Add 'resolution' and 'origin' fields to levels when using pngs.

tulku, 2019-11-28:
We would like to use this tool to create lanes and vertices over a normal 'lidar map' png.

For that our workflow would create a new level and use the map as a png, add the vertices and lanes as needed and save the project into a yaml file.

The resulting file will contain the poses of all the vertices in pixel coordinates, however we would need to convert them in map coordinates (in meters). For that we could:
* Add `measurements` and then use them as reference to calculate the resolution of the file (converting from pixels to meters), however, there is no way to encode the map origin (The 2-D pose of the lower-left pixel in the map). Also, using `measurement` can be inaccurate unless you do it very carefully.
* When converting the vertices use both the generated yaml files and a (`map_server`)[http://wiki.ros.org/map_server#YAML_format] yaml file to read the resolution and origin. This solution is not ideal as you need to keep the two files together, and it would be difficult to support multiple levels and match each level to the corresponding map yaml.

A better solution would be to add the resolution and origin properties to a level and have that saved directly on the yaml. For example:
```
levels:
  Building:
    drawing:
      filename: /full/path/map.png
      origin:
         x: 16.012
         y: 18.052
         yaw: 0
      resolution: 0.1
    elevation: 0
```
These fields could even added to the `Level->Add` dialog or use the property editor on the right to add those. The `Level->Add` dialog could also support reading the map_server yaml and read the map image, resolution and origin from that file.

Is this something that you would consider? Is there already a good mechanism to encode this information in the saved file that we did not see?

codebot, 2019-11-30:
Interesting! Thanks for the suggestion. Because our current workflow has (so far) always started with architectural drawings, which always include various explicit scale markings that are easy to capture as "measurements," we did not need a `resolution` parameter so far. It actually felt much easier to encode the scale by drawing the "measurement" edges and labeling their real-world distances as marked in the architectural floorplan. But I can understand that if you're starting from a robot-generated map, it would be much easier to just state the resolution explicitly. Hopefully this can go somewhere in `Level::calculate_scale()` ? We'd need to have some decision tree in there that would (presumably prefer explicit scale, and if it's not provided, use "measurement" edges). As for the UI, I think that would go nicely in the `level_dialog.cpp` since it's a "level-wide" parameter and I'm not sure how we would know when to populate the "property editor" in the main window, since you can't select the level, only things "within" the level.

As for `origin`, that seems fine to me. Our current workflow doesn't really care about the origin, since no humans ever see it. It seems convenient to annotate the map in pixel coordinates. Maybe the `origin` information could just be used by exporters?

I'm almost done with a reorganization of the repo so that it will contain ROS 2 packages for a `BuildingMap` message type which contains the multi-level maps that `traffic-editor` generates. As before, you'll be able to build `traffic-editor` using regular `cmake`. You'll also be able to build it using `colcon` as part of a larger ROS 2 build. We're also looking at ways to align robot-generated maps with the architectural drawings using some sort of GUI to make the process less painful. Nothing specific yet but we'll be needing some sort of mechanism for this in the next few days/weeks. Anyway just a head's up. Cheers!

codebot, 2020-02-03:
These features are hopefully all implemented now.

closed 2020-02-03 by codebot