open-rmf/rmf_traffic_editor #36: FileNotFound Error when trying to generate Gazebo world

cnboonhan94, 2020-01-11:
I am attempting to generate a gazebo world by annotation using `traffic-editor` and using `building_map_gazebo`, for a rewrite of a previous tutorial on using RMF. I have annotated a .png file using `traffic-editor`, but now when i try to run the following command from my [project root](https://github.com/cnboonhan94/fleet-adapter-tutorial/tree/rewrite) using the commands:
```
ros2 run building_map_tools building_map_gazebo ./maps/demo-world/demo-world.yaml ./maps/demo-world/demo-world.world ./maps/demo-world/models
```
i get the following error:
```

level L1 scale estimated as 0.002524245376844592
generating model of level L1 in ./maps/demo-world/models/demo-world_L1
  wrote ./maps/demo-world/models/demo-world_L1/model.config
generate_wall_visual_mesh(demo-world_L1, ./maps/demo-world/models/demo-world_L1)
  generating ./maps/demo-world/models/demo-world_L1/meshes/walls.obj
Traceback (most recent call last):
  File "/home/bhan/repos/fleet-adapter-tutorial/ros2/install/building_map_tools/lib/building_map_tools/building_map_gazebo", line 11, in <module>
    load_entry_point('building-map-tools==0.0.0', 'console_scripts', 'building_map_gazebo')()
  File "/home/bhan/repos/fleet-adapter-tutorial/ros2/install/building_map_tools/lib/python3.6/site-packages/building_map_generators/building_map_gazebo.py", line 13, in main
    g.generate_sdf(input_filename, output_filename, output_model_dir)
  File "/home/bhan/repos/fleet-adapter-tutorial/ros2/install/building_map_tools/lib/python3.6/site-packages/building_map/generator.py", line 28, in generate_sdf
    building.generate_sdf_models(output_models_dir)
  File "/home/bhan/repos/fleet-adapter-tutorial/ros2/install/building_map_tools/lib/python3.6/site-packages/building_map/building.py", line 90, in generate_sdf_models
    level.generate_sdf_model(model_name, model_path)
  File "/home/bhan/repos/fleet-adapter-tutorial/ros2/install/building_map_tools/lib/python3.6/site-packages/building_map/level.py", line 304, in generate_sdf_model
    self.write_sdf(model_name, model_path)
  File "/home/bhan/repos/fleet-adapter-tutorial/ros2/install/building_map_tools/lib/python3.6/site-packages/building_map/level.py", line 290, in write_sdf
    self.generate_walls(model_ele, model_name, model_path)
  File "/home/bhan/repos/fleet-adapter-tutorial/ros2/install/building_map_tools/lib/python3.6/site-packages/building_map/level.py", line 214, in generate_walls
    self.generate_wall_visual_mesh(model_name, model_path)
  File "/home/bhan/repos/fleet-adapter-tutorial/ros2/install/building_map_tools/lib/python3.6/site-packages/building_map/level.py", line 149, in generate_wall_visual_mesh
    with open(obj_path, 'w') as f:
FileNotFoundError: [Errno 2] No such file or directory: './maps/demo-world/models/demo-world_L1/meshes/walls.obj'
```
any idea what could be the problem?

cnboonhan94, 2020-01-11:
Seems like the issue was indeed python not creating new files if the folder did not exist prior, but `w+` seems to "weak" to create new folders. Adding the following snippet before [this line](https://github.com/osrf/traffic_editor/blob/master/building_map_tools/building_map/level.py#L149) seems to fix:
```
        if not os.path.exists(os.path.dirname(obj_path)):
            try:
                os.makedirs(os.path.dirname(obj_path))
            except OSError as exec: # Race condition guard
                if exc.errno != errno.EEXIST:
                    raise

```
If this is the ideal solution, i could make a PR.

codebot, 2020-03-05:
I think this should be now fixed upstream in the build tools. Feel free to re-open if it's still an issue after updating everything on your box

closed 2020-03-05 by codebot