open-rmf/rmf_traffic_editor #40: weird error when compiling "building_map_tools" with colcon build --symlink-install

kerliang0801, 2020-01-14:
So, I've been trying to compile with colcon build **--symlink-install**.

`building_map_tools` fails with the error message:

`
stderr: building_map_tools

error: can't copy 'building_map_generators/textures/blue_linoleum.png': doesn't exist or not a regular file
`

I'm really not sure what symlinks do to `.png`s  It somehow corrupts the pngs?

Below is the screenshot of the .pngs after I run a `colcon build --symlink-install`

[image]

Just running `colcon build` WITHOUT **--symlink-install** works fine tho.

codebot, 2020-01-14:
Interesting. I haven't ever tried to build with `--symlink-install`   I suspect the problem is with this line in the `setup.py` of that package:
https://github.com/osrf/traffic_editor/blob/master/building_map_tools/setup.py#L19

but I'm not sure what it should be in order for symlink install to work.

kerliang0801, 2020-01-14:
Yeap. It's most probably those three lines pointing to the pngs. I'll be happy to look into it when I can!

cnboonhan94, 2020-01-22:
double confirmed, symlink-install overwrites the original files with a symbolic link pointing to itself.

luca-della-vedova, 2020-01-28:
Managed to reproduce, looking into it

luca-della-vedova, 2020-02-06:
Fixed https://github.com/colcon/colcon-core/pull/301, should be available in the next colcon sync.

closed 2020-02-06 by luca-della-vedova