open-rmf/rmf_traffic_editor #44: cannot add level

cwrx777, 2020-01-31:
how to add level?
there is no menu called 'Level'.

codebot, 2020-01-31:
Sorry. Documentation needs to be updated; I'll do that right now.

There is an "Add..." button in the "Levels" tab at upper-right. If you click that, it will pop up a dialog to add a level.

codebot, 2020-01-31:
README updated in 2f14bde4bc4c2c6368de78e2cf4f9e346053f4a0

cwrx777, 2020-01-31:
[image]
I don't see any "Add..." button in the "Levels" tab at upper0right

codebot, 2020-01-31:
I see. I think you need to click "File->New..." before anything starts to happen.

The UI needs to make this more obvious.

codebot, 2020-01-31:
lol ok that causes its own problems... i'll need to make a PR to fix that :man_facepalming:

PR #45 (2020-01-31): deal more gracefully with an empty world by not crashing
fix #44
closed 2020-01-31 by codebot