open-rmf/rmf_traffic_editor #54: Key Error after modifying .building.yaml

Yadunund, 2020-02-11:
After any modification is made to the `.building.yaml` file using the traffic editor, running `building_map_nav` on same `.building.yaml' file raises the below error.

```
  File "/home/yadu/ws_rmf_demos/install/building_map_tools/lib/building_map_tools/building_map_nav", line 11, in <module>
    load_entry_point('building-map-tools==0.0.0', 'console_scripts', 'building_map_nav')()
  File "/home/yadu/ws_rmf_demos/install/building_map_tools/lib/python3.6/site-packages/building_map_generators/building_map_nav.py", line 12, in main
    g.generate_nav(input_filename, output_dir)
  File "/home/yadu/ws_rmf_demos/install/building_map_tools/lib/python3.6/site-packages/building_map/generator.py", line 40, in generate_nav
    building = self.parse_editor_yaml(input_filename)
  File "/home/yadu/ws_rmf_demos/install/building_map_tools/lib/python3.6/site-packages/building_map/generator.py", line 18, in parse_editor_yaml
    return Building(y)
  File "/home/yadu/ws_rmf_demos/install/building_map_tools/lib/python3.6/site-packages/building_map/building.py", line 9, in __init__
    self.name = yaml_node['building_name']
KeyError: 'building_name'

```

codebot, 2020-02-11:
can you see if #55 fixes it?

Yadunund, 2020-02-11:
Yes that fixes it! Thanks :smiley:

commit 0753ec2a (2020-02-11): Merge pull request #55 from osrf/update_yaml_key_names

fix #54, update yaml key names
commit 74d73ebb (2020-02-11): fix #54, update yaml key names
commit 8bcfd411 (2020-02-21): fix #54, update yaml key names
closed 2020-02-11 by codebot