open-rmf/rmf_traffic_editor #63: speed up workflow by click and drag vertex to create walls

iamironcat, 2020-02-21:
It takes a long time to create walls.

The workflow is currently:
- click edges of the walls to mark vertex
- click on wall to drag the vertex
- in some cases, the vertex can be very far apart and difficult to see when zooming out due to the small vertex
- move the vertex to adjust

To speed up creation of walls, I'd suggest something like a "path tool" from photoshop (1 min)
https://youtu.be/3JhcdVXt1X4?t=58
Where the user would click to create a vertex, drag to create a wall and so on.

The suggestion for might look something like this:
- click to create a vertex
- drag to create a wall
- click an edge while dragging to create a vertex
- esc to stop this state 
- move vertex to adjust

The only potential "issue" with this is prior planning. The user would need to plan where to draw the vertex before starting to draw. But as the path tool from photoshop, it is faster to do that than to place the vertex then draw the walls.

codebot, 2020-02-25:
#66 implements part of this improvement, where a series of clicks (no dragging required anymore) will create a chain of walls between vertices. The second half of it, creating "new" vertices with each click, will need a bit more thought.

codebot, 2020-02-25:
Done! Thanks for the suggestion :bowing_man:  it's way faster now to draw things.

closed 2020-02-25 by codebot