open-rmf/rmf_traffic_editor #70: Editor not displaying floor map immediately in new project

aaronchongth, 2020-02-28:
After following the steps listed on the `README.md` to start a new project, adding a floor plan using `Levels->Add`, for example [office.png(https://github.com/osrf/rmf_demos/blob/master/rmf_demo_maps/maps/office/office.png),  the floor plan does not show up, nor does the level show up in the levels list.

Only after saving the project (`Ctrl + S` ), exiting `traffic_editor`, and opening `traffic_editor` again, then will the correct floor plan and level be display.

aaronchongth, 2020-02-28:
after inserting floor plan,
[image]

console log of new project, adding floor plan, saving via `Ctrl + S`,
https://gist.github.com/aaronchongth/e39a07ed3fdd57867e6bc395d5377ad7#file-traffic_editor-new-project-adding-floor-plan

after saving, closing and reopening floor plan,
[image]

codebot, 2020-02-28:
Thanks for the detailed report. I can replicate it on my machine as well. Fix in progress...

PR #71 (2020-02-28): Fix initial creation workflow
Fix #70
closed 2020-02-28 by codebot