open-rmf/rmf_traffic_editor #99: Doors with negative motion direction not behaving correctly

aaronchongth, 2020-03-20:
simulated doors/plugins do not behave as drawn on `traffic_editor`, when `motion_direction` is toggled to `-1`

Fixes will be added to `building_map_tools` and `door.cpp` to make simulated doors behave as expected

codebot, 2020-03-20:
related to https://github.com/osrf/traffic_editor/pull/97/files   ? I suppose that logic needs to incorporate the `motion_direction` sign somewhere and branch based on its sign.

aaronchongth, 2020-03-20:
yup! I was expecting users to redraw the door in the opposite directions (changing start and end of an edge) instead of changing the motion direction, that was careless of me. 

I'll try to verify https://github.com/osrf/traffic_editor/pull/100 asap

closed 2020-03-22 by aaronchongth