open-rmf/rmf_traffic_editor #101: Hinged doors should have an extra link at joints, so doors don't clip into walls

aaronchongth, 2020-03-22:
as mentioned in https://github.com/osrf/rmf_demos/pull/42

codebot, 2020-03-23:
I suppose we could generate an extra little bit of wall automatically, but since each building (and sometimes even each door) has a somewhat different amount of "extra wall" between it and the perpendicular wall, I'd suggest that we draw it explicitly on the map. But we can definitely move the hinge point for swing doors "inside" the door threshold line segment a bit, so that they don't always rotate half of the door inside the wall when it's open 90 degrees :heavy_check_mark:

aaronchongth, 2020-03-23:
I tried out the few options and believe that moving the joint axis inwards by a fixed offset is definitely the most elegant solution, we will then have no need for extra links or walls, and we also won't need to update all the older `.building.yaml`

The offset is currently 3 times the door's thickness,
[image]

aaronchongth, 2020-03-23:
for completeness sake, here is a screenshot of how it looks like when its fully opened 90 degrees.
 
[image]