open-rmf/rmf_traffic_editor #146: Fail to generate cropped thumbnail image

MakinoharaShouko, 2020-06-16:
I tried to use thumbnail generator from the traffic editor to add some gazebo models. When I ran
```
./scripts/crop.py model_list.yaml -o images/cropped -g images/green -w images/white
```
I got the following message
```
./scripts/crop.py model_list.yaml -o images/cropped -g images/green -w images/white
generating images/cropped/Nate Koenig/cabinet.png
green image size: 2000x2000
cropped to 124x124
```
It seems a cropped thumbnail image should have been generated, but when I went to /images/cropped/ the folder was empty. By the way I was able to generate images with green and white backgrounds. I am wondering what went wrong for the cropping step

MakinoharaShouko, 2020-06-16:
Fixed this by removing the author name from model_list.yaml

aaronchongth, 2020-06-16:
We do have plans on working with namespaced models once we get the Fuel pipeline running, thanks for testing this! We can keep this issue open until we are done with the migration.

methylDragon, 2020-06-29:
This should be fixed already, with the merge for https://github.com/osrf/traffic_editor/pull/147

closed 2020-06-29 by methylDragon