open-rmf/rmf_traffic_editor #163: pit_crew fails when ~/.gazebo or ~/.ignition not present

Briancbn, 2020-06-29:
### Bug Report
@methylDragon I think this is a small bug to fix
I notice that when I build traffic editor in my pipeline the first time without downloading any models (hence, no `~/.gazebo` or `~/.ignition`). It will give me the following error.
```
Starting >>> test_maps
--- stderr: test_maps                              
BUILDING WORLDFILE WITH COMMAND: ros2 run building_map_tools building_map_generator gazebo /home/.../rmf_ws/src/traffic_editor/test_maps/maps/door_madness/door_madness.building.yaml /home/.../rmf_ws/build/test_maps/maps/door_madness/door_madness.world /home/.../rmf_ws/build/test_maps/maps/door_madness/models
Traceback (most recent call last):
  File "/home/.../rmf_ws/install/building_map_tools/lib/building_map_tools/building_map_generator", line 11, in <module>
    load_entry_point('building-map-tools==0.0.0', 'console_scripts', 'building_map_generator')()
  File "/home/.../rmf_ws/install/building_map_tools/lib/python3.6/site-packages/building_map_generator/building_map_generator.py", line 51, in main
    lower=True
  File "/home/.../rmf_ws/install/building_map_tools/lib/python3.6/site-packages/pit_crew/pit_crew.py", line 153, in get_missing_models
    use_dir_as_name=use_dir_as_name, ign=ign
  File "/home/.../rmf_ws/install/building_map_tools/lib/python3.6/site-packages/pit_crew/pit_crew.py", line 245, in get_local_model_name_tuples
    "Path given must be a directory that exists!"
```
I can work around it by setting NO_DOWNLOAD_MODELS or create a `~/.gazebo` directory, but I think it's best this can be addressed properly.

methylDragon, 2020-06-29:
Hi @Briancbn , let me check if a fix breaks anything else.

methylDragon, 2020-06-29:
Fixed in https://github.com/osrf/traffic_editor/pull/165

(:

Briancbn, 2020-06-30:
Thanks works now.

closed 2020-06-30 by Briancbn