open-rmf/rmf_traffic_editor #177: Failed to load system plugin

MakinoharaShouko, 2020-07-04:
I am now getting the following errors when launching rmf_demos with ignition. Because of these errors, I can now only start the 3D scenario but cannot send requests. I don't know what is causing the problem, since the traffic_editor was built and the files reporting errors are present.
```
[ign-8] [Err] [SystemLoader.cc:101] Failed to load system plugin [slotcar] : system not found in library  [libslotcar.so] from path [/home/makinohara/rmf_demos_ws/install/building_ignition_plugins/lib/libslotcar.so].
[ign-8] [Err] [SystemLoader.cc:101] Failed to load system plugin [slotcar] : system not found in library  [libslotcar.so] from path [/home/makinohara/rmf_demos_ws/install/building_ignition_plugins/lib/libslotcar.so].
[ign-8] [Err] [SystemLoader.cc:101] Failed to load system plugin [door] : system not found in library  [libdoor.so] from path [/home/makinohara/rmf_demos_ws/install/building_ignition_plugins/lib/libdoor.so].
[ign-8] [Err] [SystemLoader.cc:101] Failed to load system plugin [door] : system not found in library  [libdoor.so] from path [/home/makinohara/rmf_demos_ws/install/building_ignition_plugins/lib/libdoor.so].
[ign-8] [Err] [SystemLoader.cc:101] Failed to load system plugin [door] : system not found in library  [libdoor.so] from path [/home/makinohara/rmf_demos_ws/install/building_ignition_plugins/lib/libdoor.so].
```
I am using feature/add_ignition_support branch for rmf_demos and feature/ign_lift_plugin for traffic_editor. Both are up to date and built

codebot, 2020-07-13:
@luca-della-vedova  any tips?

luca-della-vedova, 2020-07-14:
I would probably try to use the ign_plugins branch first since it's a bit more stable, and wait for the lift pr to be merged before trying to do multilevel simulations.

MakinoharaShouko, 2020-07-14:
I switched to ign_plugins and built again, the problem persists
Here is the warning from `building_ignition_plugins` I got during building, I don't know if it is related (I think I have seen it in buildings that did not have the problem)
```
--- stderr: building_ignition_plugins       
/home/makinohara/rmf_demos_ws/src/traffic_editor/building_sim_plugins/building_ignition_plugins/src/door.cpp: In member function ‘virtual void building_ignition_plugins::DoorPlugin::Configure(const Entity&, const std::shared_ptr<const sdf::v9::Element>&, ignition::gazebo::v4::EntityComponentManager&, ignition::gazebo::v4::EventManager&)’:
/home/makinohara/rmf_demos_ws/src/traffic_editor/building_sim_plugins/building_ignition_plugins/src/door.cpp:61:48: warning: unused parameter ‘_eventMgr’ [-Wunused-parameter]
     EntityComponentManager& ecm, EventManager& _eventMgr) override
                                  ~~~~~~~~~~~~~~^~~~~~~~~
---
```

luca-della-vedova, 2020-07-14:
The warning is unrelated, I'm really not sure what is happening, can you post the result of
`which ign`
and
`ign gazebo --version`
Also might be good to go through all the repos in the ignition workspace and do a git pull to make sure they are up to date

MakinoharaShouko, 2020-07-14:
> The warning is unrelated, I'm really not sure what is happening, can you post the result of
> `which ign`
> and
> `ign gazebo --version`
> Also might be good to go through all the repos in the ignition workspace and do a git pull to make sure they are up to date

I just the checked that all the repos in `ws_dome` were on the correct branch and pulled the latest version. The problem still exists after building.
Here is the result for the two commands you mentioned
```
makinohara@makinohara-Predator-G3-571:~/ws_dome$ which ign
/home/makinohara/ws_dome/install/bin/ign
makinohara@makinohara-Predator-G3-571:~/ws_dome$ ign gazebo --version
Ignition Gazebo, version 2.20.1
Copyright (C) 2018 Open Source Robotics Foundation.
Released under the Apache 2.0 License.
```

luca-della-vedova, 2020-07-14:
> makinohara@makinohara-Predator-G3-571:~/ws_dome$ ign gazebo --version
> Ignition Gazebo, version 2.20.1
> Copyright (C) 2018 Open Source Robotics Foundation.
> Released under the Apache 2.0 License.
> ```
If you are building from source the output should be:
```
Ignition Gazebo, version 4.0.0~pre1
Copyright (C) 2018 Open Source Robotics Foundation.
Released under the Apache 2.0 License.
```
Make sure you are on the right ign-gazebo branch (master) and that you sourced your dome_ws install.bash file

MakinoharaShouko, 2020-07-19:
So I removed the libignition-gazebo2-dev and rebuilt ignition dome from source, now I am encountering the problem of gazebo not available on my computer.
[image]
I am confused why this happens, since the building of ignition-gazebo4 finished with no problem and I have also sourced the setup.bash file.

luca-della-vedova, 2020-07-19:
This is very puzzling, I wonder if anything changes if you take away the
symlink install option?

MakinoharaShouko, 2020-07-19:
No it is still the same output
I also tried binary install of libignition-gazebo4-dev but kept getting package could not be located. I have added package repos according to [this document](https://github.com/ignitionrobotics/ign-gazebo).

luca-della-vedova, 2020-07-27:
Apologies forgot about this, I noticed from your screenshot that it was not a clean build. Can you try to remove all the build folders (build, install and log) and try to rebuild just doing:

`colcon build --merge install`

MakinoharaShouko, 2020-07-28:
I tried to remove everything and then build again. It is still not working

I got the full output for building
```
Starting >>> ignition-cmake2
Starting >>> ignition-tools
Finished <<< ignition-cmake2 [1.83s]                                        
Starting >>> ignition-math6
--- stderr: ignition-tools                                            
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
---
Finished <<< ignition-tools [2.53s]
[Processing: ignition-math6]                              
[Processing: ignition-math6]                                      
Finished <<< ignition-math6 [1min 14s]                                 
Starting >>> ignition-common3
Starting >>> ignition-msgs5
Starting >>> sdformat9
[Processing: ignition-common3, ignition-msgs5, sdformat9]                                                                                     
[Processing: ignition-common3, ignition-msgs5, sdformat9]                                                                                                 
[Processing: ignition-common3, ignition-msgs5, sdformat9]                                                                                                    
[Processing: ignition-common3, ignition-msgs5, sdformat9]                                                                                                    
[Processing: ignition-common3, ignition-msgs5, sdformat9]                                                                                                    
[Processing: ignition-common3, ignition-msgs5, sdformat9]                                                                                                    
Finished <<< ignition-common3 [3min 17s]                                                                                                                        
Starting >>> ignition-plugin1
[Processing: ignition-msgs5, ignition-plugin1, sdformat9]                                                                                               
[Processing: ignition-msgs5, ignition-plugin1, sdformat9]                                                                                                   
Finished <<< ignition-plugin1 [1min 13s]                                                                                                                        
Starting >>> ignition-rendering4
Finished <<< ignition-msgs5 [4min 59s]                                                                                                                        
Starting >>> ignition-transport8
Starting >>> ignition-fuel_tools4
[Processing: ignition-fuel_tools4, ignition-rendering4, ignition-transport8, sdformat9]                                                                                                            
[Processing: ignition-fuel_tools4, ignition-rendering4, ignition-transport8, sdformat9]                                                                                                                  
[Processing: ignition-fuel_tools4, ignition-rendering4, ignition-transport8, sdformat9]                                                                                  
[Processing: ignition-fuel_tools4, ignition-rendering4, ignition-transport8, sdformat9]                                                                                  
[Processing: ignition-fuel_tools4, ignition-rendering4, ignition-transport8, sdformat9]                                                                                  
--- stderr: ignition-fuel_tools4                                                                                                                                         
CMake Warning (dev) at /home/makinohara/ws_dome/install/share/cmake/ignition-cmake2/cmake2/IgnUtils.cmake:378 (message):
   -- THIS MESSAGE IS INTENDED FOR IGNITION-FUEL_TOOLS AUTHORS --
      (IF YOU SEE THIS, PLEASE REPORT IT)

  Could not find pkg-config information for YAML.  It was not provided by the
  find-module for the package, nor was it explicitly passed into the call to
  ign_find_package(~).  This is most likely an error in this project's use of
  ign-cmake.
Call Stack (most recent call first):
  CMakeLists.txt:41 (ign_find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< ignition-fuel_tools4 [2min 35s]
Finished <<< sdformat9 [8min 3s]                                                                                                                                      
Starting >>> ignition-physics2
[Processing: ignition-physics2, ignition-rendering4, ignition-transport8]                                                                                              
[Processing: ignition-physics2, ignition-rendering4, ignition-transport8]                                                                                                   
[Processing: ignition-physics2, ignition-rendering4, ignition-transport8]                                                                                                     
[Processing: ignition-physics2, ignition-rendering4, ignition-transport8]                                                                                                     
[Processing: ignition-physics2, ignition-rendering4, ignition-transport8]                                                                                                     
[... cut ...]
Call Stack (most recent call first):
  CMakeLists.txt:49 (ign_find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< ignition-launch3 [19.9s]

Summary: 15 packages finished [24min 23s]
  4 packages had stderr output: ignition-fuel_tools4 ignition-launch3 ignition-rendering4 ignition-tools
```
The warnings are not from ignition-gazebo. I am not sure if they are related. Additionally, I built nvidia optix on my machine, but the warning of missing optix still exists

luca-della-vedova, 2020-07-28:
All the warnings (including Optix) are unrelated don't worry about them.

I am honestly very puzzled, other than making sure you are on the right branches and pulling the latest version for each of the repositories I'm not sure how to proceed. I have never seen / heard of anyone having such an issue before. I wonder if there is some conflict with some other workspace you are sourcing (did you add a lot of things to your .bashrc?)