open-rmf/rmf_traffic_editor #178: Feature: View->Fit

codebot, 2020-07-05:
This feature doesn't currently work. It would be helpful when (for example) defining fiducials, because the coordinate system can change and the previously-saved X,Y,scale coordinates can be nonsense.

txlei, 2021-03-24:
hi, for some strange reason my traffic-editor is not displaying anything. I did a terrible mistake of not using a mouse and ended up zooming in too much. Tried to reinstall traffic editor and also qt related libraries to reset the view config if any. Any suggestion? 
[image]
[image]

codebot, 2021-03-24:
The settings (including last-used viewport center and zoom factor) should be saved in `~/.config/open-robotics/traffic-editor.conf`

It's just a text file in Qt's QSettings format. In that file, there should be a line like `viewport_scale=` that you can just delete. If it doesn't find that key in the QSettings file, it should just start up with a scale of 1.0.

codebot, 2021-04-07:
fixed in #318 , hopefully it's impossible to get in an extremely-zoomed-out configuration now, and even if it is, the Zoom...Reset menu action should recover it now.

closed 2021-04-07 by codebot