open-rmf/rmf_traffic_editor #190: Thumbnail generator error with image cropping

ngwk1, 2020-07-15:
I have received an error for using the script `crop.py`:

```
Traceback (most recent call last):
  File "./scripts/crop.py", line 149, in <module>
    white_img_cropped = crop(green_img, white_img)
  File "./scripts/crop.py", line 32, in crop
    mask, cv2.RETR_EXTERNAL, cv2.CHAIN_APPROX_NONE)
ValueError: need more than 2 values to unpack
```

I fixed it by changing the number of outputs from:
`   _ ,  contours, _ = cv2.findContours( mask, cv2.RETR_EXTERNAL, cv2.CHAIN_APPROX_NONE) `
to 
`    contours, _ = cv2.findContours(mask, cv2.RETR_EXTERNAL, cv2.CHAIN_APPROX_NONE)`

aaronchongth, 2020-07-16:
thanks for finding that issue, this is probably due to different versions of OpenCV, while the version shipped with ROS prefers the latter version. I'll make the necessary fixes, thanks!

aaronchongth, 2020-07-17:
@ngwk1 alternative, you could take a look at this new pipeline, it should work as expected. if you find any issues, feel free to reach out
https://github.com/osrf/traffic_editor/pull/191

tanyouliang95, 2020-07-20:
Yes, we have a replacement for the existing `thumbnail_generator`. Can close this issue for now. Thanks for raising this :+1:

closed 2020-07-20 by tanyouliang95