open-rmf/rmf_traffic_editor #193: Thumbnail generator: Unable to load Ogre Resources

Briancbn, 2020-07-21:
A follow up on #190 
First of all, thanks @tanyouliang95 for fast fix in #191 .
I am facing this error when using both the python script and the plugins directly running with `--verbose`
```
[Err] [RenderEngine.cc:538] EXCEPTION: Unable to load Ogre Resources.
```
Most other models are fine, but those using `<script>` for `<material>` will fail. 
- Gzserver version: 9.13.1
- OpenCV version: 3.2.0
- Software used to generate the COLLADA file: 3ds max

I have included an example link to download (Not done by me). It works fine in Gazebo World
[Here is the link](https://drive.google.com/file/d/16qdjc2DFH9E9JpqjWwmLD62l4JpPqkS3/view?usp=sharing)

tanyouliang95, 2020-07-22:
Hi @Briancbn, thanks for trying out. I have the same env as urs yet I'm not able to recreate the described issue. Here is a generated thumbnail with the provided model. Besides, I also had a successful test with a similar [model](https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Oak%20tree), which consists of `<material>` field. Maybe upgrade to the latest `ogre`? Let me know again. 
[image]

On regards to the segfault during destruction, this will not hinder the creation of the thumbnail. I was able to recreate the issue only when I'm running it on a different PC with the same distribution. This is odd, I will also try to figure this out for now.

Briancbn, 2020-07-26:
Thanks @tanyouliang95 for the prompt reply. It seems that I didn't set the OGRE_RESOURCE_PATH before running the script. In the end, simply sourcing the gazebo setup script solve the problem
```bash
. /usr/share/gazebo/setup.sh
```
The segfault still persists for me tho.

tanyouliang95, 2020-07-28:
Since thumbnails can be generated. I have closed this issue. The seg fault will soon be addressed separately.

closed 2020-07-28 by tanyouliang95