open-rmf/rmf_traffic_editor #204: Reversible option in slotcar plugin

Briancbn, 2020-08-04:
# Feature Request
Slotcar plugin can be told when the vehicle is reversible, https://github.com/osrf/rmf_core/issues/144.
# Current behavior
Version used:
**rmf_core**: 1.0.2
**traffic editor**: 1.0.0
**rmf_demos**: https://github.com/osrf/rmf_demos/commit/dbe78f45973989283bac2b2451f93597a879094c

I simply added this line under [rmf_demos:magni_adapter.launch.xml](https://github.com/osrf/rmf_demos/blob/master/demos/launch/include/adapters/magni_adapter.launch.xml) and tested in simulation

```xml
      <arg name="reversible" value="false" />
```

The behaviour can be reproduced running [rmf_demos:office_loop_conflict.launch.xml](https://github.com/osrf/rmf_demos/blob/master/demos/launch/include/office/office_loop_conflict.launch.xml)

Although the fleet does not normally go back, but during the replanning and negotiation when conflicts are detected, the vehicle still reverse constantly.

mxgrey, 2020-08-20:
For reference, this needs to be fixed in two places:

1. [Here](https://github.com/osrf/traffic_editor/blob/master/building_sim_plugins/building_plugins_common/src/slotcar_common.cpp#L249) in the slotcar plugin. If the vehicle is not reversible, then we shouldn't use the `&dir` parameter.
2. Somewhere in the `traffic_editor` there should be a way to specify whether a vehicle is reversible.

One issue that may arise is that the reversibility parameter needs to be kept consistent between the plugin and the launch file, otherwise this kind of issue will persist. The current situation is a multiple-source-of-truth anti-pattern. Unfortunately I don't have an easy solution to suggest for this at the moment.

mxgrey, 2021-05-17:
I believe this issue can be closed by https://github.com/open-rmf/rmf_traffic_editor/pull/254

With that PR, the slotcar will face in the requested direction when moving, so it doesn't need to worry about whether or not the robot is reversible.

closed 2021-05-17 by mxgrey