open-rmf/rmf_traffic_editor #219: Deselecting a floor in Gazebo does not hide the models on the floor

mxgrey, 2020-08-20:
As shown in the screenshot below, when you toggle off the visibility for a floor, the models on that floor remain visible. It would probably be better if those could also be turned off, so we don't see mysterious floating beings.

[image]

codebot, 2020-08-20:
Ah! Indeed. That's not super great. Thank you for finding this.

codebot, 2020-08-20:
There are many ways to go about this, but at the moment it seems the simplest way would be to look at the Z height of the floor that is being toggled, and then use a heuristic window around that Z height (say, -0.2 through +2.5 meters) and toggle all models (static only?) that lie within that Z window. Sound OK?

mxgrey, 2020-08-20:
That sounds okay for a first-order approach. I think there could be some loose ends if an object has a wonky origin (leading to false positives or false negatives), but that simple approach should cover 99% of cases.

codebot, 2020-08-20:
Cool. Then we can avoid (at least for now) adding some sort of metadata to the automatically-generated models and trying to fish it out in the plugin later. Using the Z window feels like a minimal first attempt, and we can see how it goes.

Yadunund, 2020-09-01:
@Kevinskwk has proposed a method where a list of static model names is appended as sdf subelements of the `toggle_floors` plugin in the `.world` file. https://github.com/osrf/traffic_editor/pull/226
When a level visibility is toggled, the plugin turns off the visibility of appropriate models from this list. On the pro side, it avoids having to manually define Z height boundaries but on the cons side, it increases the verbosity of the `.world` file. Any feedback on this approach?

mxgrey, 2020-09-01:
Sounds like a good approach. The last question would be whether we want to see robots or non-static models floating, and what should we do about it if we don't want that?

Yadunund, 2020-09-02:
We can extend the same behavior to non-static models such as doors. For robots things are trickier as they can move between levels. One option would be to subscribe to RobotState/FleetState messages and filter out robots whose reported levels match the toggle selection.

luca-della-vedova, 2022-04-06:
This was done some time ago in Gazebo and now also in Ignition, closing.

closed 2022-04-06 by luca-della-vedova