open-rmf/rmf_traffic_editor #240: 3D waypoint position

MakinoharaShouko, 2020-10-02:
Currently, the nav_graph waypoints contain only x & y information. Considering that there are now many multi-level maps, would it be reasonable to add the z information into the nav_graph as well?

codebot, 2020-10-02:
I just had a look at the `clinic` nav graphs generated by a build of the `rmf_demo_maps` package. It looks like the vertices in the data structure are already children of their parent level, so the nav_graph waypoints are effectively stored as {level_name, x, y}. Since RMF currently deals with ground robots that have to use elevators to navigate between floors, that sounds reasonable to me. Are you thinking of drones or other types of domains where full 3d navigation is required, rather than "a stack of flatlands" style navigation?

MakinoharaShouko, 2020-10-02:
I was reading the `rmf_core` along with the `traffic_editor`. It seems to me that the euclidean expander was only calculating based on the x and y values. I am wondering if the z value could be added to the nav_graph to improve the route planning.

mxgrey, 2021-01-07:
This issue is resolved by https://github.com/osrf/rmf_core/pull/243. The new [Euclidean Distance Heuristic](https://github.com/osrf/rmf_core/blob/master/rmf_traffic/src/rmf_traffic/agv/planning/EuclideanHeuristic.cpp) of the planner will account for the need to use lifts when moving between floors. The heuristic is now calculated by doing a quick search for the sequence of lifts that offers the shortest lift-to-lift distance when moving from the initial floor to the destination floor.

closed 2021-01-07 by mxgrey