open-rmf/rmf_traffic_editor #246: Generated world file missing <xmlns:ns="rmf"> from template

Briancbn, 2020-10-13:
I tried to parse the generated gazebo world file through ElementTree and got this error
```bash
xml.etree.ElementTree.ParseError: unbound prefix
```
My code is
```python3
from xml.etree import ElementTree
tree = ElementTree.parse("<path-to>/<name>.world")
```
It doesn't recognize the `rmf` namespace. Although the namespace line exists in [template file gz_world.sdf](https://github.com/osrf/traffic_editor/blob/3b724bfc7f918e008ebec8ae317da782994301ab/building_map_tools/building_map/templates/gz_world.sdf#L3), but it doesn't appears in my generated world file (an example will be `build/test_maps/maps/door_madness.world`).

This doesn't break Gazebo or Ignition, tinyxm2 seems to ignore this, but it's a small bug that prevents anyone who wants to do some post processing of the generated map files using etree like me.

mrushyendra, 2020-10-14:
Turns out ElementTree's `parse()` function does not store the xmlns attribute in an element's `attrib` dict, which means it disappears during the world generation process, because we are parsing the template, and then generating the sdf file from it.

(Related: https://stackoverflow.com/questions/1953761/accessing-xmlns-attribute-with-python-elementree, https://stackoverflow.com/questions/35574815/how-to-force-elementtree-to-keep-xmlns-attribute-within-its-original-element)

Briancbn, 2020-10-16:
Fixed by #253

closed 2020-10-16 by Briancbn