open-rmf/rmf_traffic_editor #273: How do I run the traffic map generator?

Marcusttk, 2020-12-14:
So I followed this page, the part where "To auto-generate a Gazebo simulation world, the executable takes in the command argument gazebo along with others described below"

In my ubuntu 20.04, I went to the traffic_editor directory and building_map_generator directory and run the building_map_generator command, but it says building_map_generator: command not found.

I already have the traffic editor setup, based on the instructions in the readme. What else am I missing to generate the building map?

codebot, 2020-12-14:
Can you copy paste the exact terminal input and output? That way we’ll be
better able to diagnose what’s happening and make suggestions. Thanks!

Marcusttk, 2020-12-14:
marcus@hp:~/colcon_workspace$ building_map_generator gazebo test
building_map_generator: command not found

^ For the exact input and output.

Heres a screenshot of my terminal.

https://imgur.com/eEUeGqV

codebot, 2020-12-14:
I see. So unlike the `traffic-editor` executable, the `building_map_generator` program is not placed on `$PATH` and needs to be run via `ros2 run`:

```
ros2 run building_map_tools building_map_generator [ARGUMENTS...]
```

Marcusttk, 2020-12-15:
Thanks. That works!

closed 2020-12-15 by Marcusttk