open-rmf/rmf_traffic_editor #283: Unable to delete unwanted levels in traffic editor

destkk, 2021-01-20:
Are there any other ways that I could delete the unwanted levels in traffic editor GUI? Currently the "DELETE" function is not working nor the "EDIT" window has a delete function.

I could manually delete it from the folder after generating the files but it's not the ideal method.

[image]

codebot, 2021-01-20:
Currently there is no way to do this in the GUI. The fastest way currently is to go into the `PROJECT_NAME.building.yaml` file and remove the sub-key and corresponding map under the `levels` key by hand in the YAML, then re-start traffic-editor.

In general we have attempted to keep the YAML format simple enough to allow such hand-editing as a fallback from the GUI editor.

codebot, 2021-01-20:
I'll leave this issue open because this would be a good feature to add in the future. Thank you for suggesting it.