open-rmf/rmf_traffic_editor #297: Unwanted lanes added when adding a new lane

Yadunund, 2021-02-22:
Running into this issue when creating lanes. Best explained by the video attached.
In the video below, I first add a lane between the waypoint on the left and the one on the right. I then hit `Esc` and then try adding a lane from the bottom waypoint to the left one. But when I click the bottom waypoint, a lane from the right waypoint to the bottom one gets added. It's almost as if the `add_lane` tool does not reset when `Esc` is hit. Only the rendering of the blue preview resets.

https://user-images.githubusercontent.com/13482049/108651285-66c99680-74fc-11eb-949d-deeb8088f58a.mp4

codebot, 2021-02-22:
Thanks for the detailed report and screen capture. I also just noticed this when doing the fix for #295 . I'll dive into this now.

codebot, 2021-02-22:
I believe this issue (and a related potential crash) are resolved here: https://github.com/osrf/traffic_editor/pull/298

commit 461e1c01 (2021-02-22): correctly exit add-edge mode, fix #297
commit 80012185 (2021-02-22): Merge pull request #298 from osrf/bug/add_lane_reset_tool

correctly exit add-edge mode, fix #297
closed 2021-02-22 by codebot