open-rmf/rmf_traffic_editor #305: clicked unhandled type: 7 when adding new layer

Achllle, 2021-03-16:
I am able to use all functions of `traffic_editor` (adding lanes, adding levels, importing png, ..) properly except the `Add` button in the `layers` window. When clicked, nothing happens and I get the following log:

```
clicked unhandled type: 7
```
I'm following the instructions in [deriving-coordinate-space-transforms](https://osrf.github.io/ros2multirobotbook/traffic-editor.html#deriving-coordinate-space-transforms)

destkk, 2021-03-29:
Hi Achille I am also having the same issue. Did you manage to resolve this?

codebot, 2021-03-29:
Adding layers and adding levels are likely to be unrelated issues. @destkk we'll work with the adding-level issue in the other ticket you opened.

codebot, 2021-03-29:
Hopefully fixed in #313

closed 2021-03-30 by codebot

Achllle, 2021-03-30:
@codebot Tried the latest master including these commits and I am still not able to use the 'add' and 'delete' buttons in the lower right of the window. I'm only able to hide a layer currently. Creating a new layer with the same name doesn't overwrite the old one.
Perhaps a separate issue but related: png files in layers also don't update when you select a different image.

codebot, 2021-03-31:
Thanks for nothing these items. I'll create three new tickets for them, so they aren't forgotten:

Add/Delete buttons: if I understand the issue correctly, you're asking about the add/delete buttons in the extreme lower-right of the window (? I could be mis-understanding). The extreme lower-right add/delete buttons are for the properties table, so they are used to add properties to vertices (not layers) for things like setting a vertex name, specifying if it's a charger, etc. We should change the text of those buttons to make that clear. I'll add a ticket for that.

Updating layer images: good point, thanks for catching this. I suspect it would update after a save-exit-open cycle, but that's not great. To date, I suspect we haven't noticed this because the image filenames change very infrequently (if ever) in our use. I'll add a ticket to re-load the image when the filename changes in the `Edit...` dialog, thanks for noticing this.

Creating a new layer with the same name: internally the layers are referenced by index in the layer table, so it's possible to have multiple layers with the same name. What behavior would you expect if a new layer is created with the same name? (dialog warning box, etc.?)

Weird behavior: ahhh, I see. I can replicate it now -- if multiple layers are created with the same name, the `Edit...` button next to them seems to always pop up an edit dialog for the first layer of that name. I'll create a ticket for this. Thanks for catching it.

commit 7999f961 (2021-03-30): Bug/add layer button not visible in new building (#313)

* avoid crash in empty crowdsim save routine
* immediately render new layers. fix #305