open-rmf/rmf_traffic_editor #311: Adding a level does not respond

destkk, 2021-03-29:
Ubuntu 20.04

Hi there. I have tried by adding a new level in traffic editor and importing a png file and giving it a level name. After adding it the image does not appear. Previously it was working fine I could open the project.yaml file from rmf_demos and it is working well. Out of a sudden it does not work for any of the project files.
```
changing directory to [/home/rosuer/rmf_demos_ws/install/rmf_demo_maps/share/rmf_demo_maps/hotel]
Building::load_yaml_file(hotel.building.yaml)
changing directory to [/home/rosuer/rmf_demos_ws/install/rmf_demo_maps/share/rmf_demo_maps/hotel]
parsing level [L1]
parsing level [L2]
used 2 measurements to estimate meters/pixel as 0.05000
used 2 measurements to estimate meters/pixel as 0.05000
transform 0->1: scale = 1.00000 translation = (-54.00, 0.00)
transform 1->0: scale = 1.00000 translation = (54.00, 0.00)
project does not have a sim plugin
```

Does anyone know how to resolve this issue? Is it because the proejct does not have sim plugin?

codebot, 2021-03-29:
Sim plugin is unrelated and not an issue here. Just checking, so we can home in on the issue: "what if" you add a level, exit, and then re-open traffic-editor -- does that make the level background image appear?

codebot, 2021-03-29:
Ohhh -- it looks like you haven't built traffic-editor from source in a while (few weeks?) since your screenshot shows a `Project` menu. It should show a `Building` menu now in the upper-left. Can you update your local checkout and rebuild from source?

destkk, 2021-03-29:
> Ohhh -- it looks like you haven't built traffic-editor from source in a while (few weeks?) since your screenshot shows a `Project` menu. It should show a `Building` menu now in the upper-left. Can you update your local checkout and rebuild from source?

Hi there. May I know how do I rebuild from source and what I should do with the old files? Sorry I'm a newbie in this.

codebot, 2021-03-29:
How did you currently install the `traffic_editor` version that you are running in the screenshot above?

destkk, 2021-03-29:
> How did you currently install the `traffic_editor` version that you are running in the screenshot above?

Via the rmf_demos installation.md https://github.com/osrf/rmf_demos/blob/master/docs/installation.md

destkk, 2021-03-29:
I have tried reinstalling the rmf_demos and now it shows this error, 

```
Package 'building_map_tools' not found
make[2]: *** [CMakeFiles/generate_battle_royale_nav_graphs.dir/build.make:61: maps/battle_royale/nav_graphs/phony] Error 1
make[1]: *** [CMakeFiles/Makefile2:211: CMakeFiles/generate_battle_royale_nav_graphs.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Package 'building_map_tools' not found
Package 'building_map_tools' not found
```
Is it because it has been updated to rmf_building_map_tools?

codebot, 2021-03-29:
yeah, we are chatting now to try to figure out the easiest way forward.

For sure you should be able to uninstall the `apt` packages that were installed using the instructions you sent, and then following the from-source build instructions here: https://github.com/open-rmf/rmf

there is a lot in flight currently as we have "graduated" RMF into its own `open-rmf` organization and been changing names so they can be released in the public build farm. The from-source build should definitely work, though (that's what we're all using), but if there are problems please let us know. We are working to update all the binary packages now also.

destkk, 2021-03-29:
> yeah, we are chatting now to try to figure out the easiest way forward.
>
> For sure you should be able to uninstall the `apt` packages that were installed using the instructions you sent, and then [...]

I have build from source based on the instructions in  https://github.com/open-rmf/rmf and I am still receiving the error as of the above building_map_tools.. what should i uninstall in the apt packages then?

marcoag, 2021-03-29:
is the package `ros-foxy-building-map-tools` installed and the setup.bash sourced?

destkk, 2021-03-29:
> is the package `ros-foxy-building-map-tools` installed and the setup.bash sourced?

Hi there. I have not installed `ros-foxy-building-map-tools `before. And yes setup.bash is sourced. I tried `sudo apt-get install ros-foxy-building-map-tools `and it is unable to locate package?

destkk, 2021-03-29:
Previously I followed rmf_demos instructions https://github.com/osrf/rmf_demos/blob/master/docs/installation.md
Now i have tried to reinstall and build based on https://github.com/open-rmf/rmf and I received the same errors.

I have not faced this issue before during the initial installation few weeks back.

codebot, 2021-03-29:
what does your console respond to `which traffic-editor`  

if you have built `traffic-editor` from source, we need to make sure that your console is running that from-source build when you type `traffic-editor`, as opposed to the version installed via `apt`

destkk, 2021-03-29:
I have installed `ros-foxy-building-map-tools etc. ` Now i'm facing this error..

```
make[2]: *** [CMakeFiles/rmf_traffic_ros2.dir/build.make:245: CMakeFiles/rmf_traffic_ros2.dir/src/rmf_traffic_ros2/schedule/Writer.cpp.o] Error 1
make[2]: *** [CMakeFiles/rmf_traffic_ros2.dir/build.make:219: CMakeFiles/rmf_traffic_ros2.dir/src/rmf_traffic_ros2/schedule/Node.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:194: CMakeFiles/rmf_traffic_ros2.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rmf_traffic_ros2 [8.34s, exited with code 2]

Summary: 41 packages finished [13.0s]
  1 package failed: rmf_traffic_ros2
  1 package had stderr output: rmf_traffic_ros2
  7 packages not processed

```

> what does your console respond to `which traffic-editor`
> 
> if you have built `traffic-editor` from source, we need to make sure that your console is running that from-source build when you type `traffic-editor`, as opposed to the version installed via `apt`

how do I know the traffic-editor is running from which source build?

codebot, 2021-03-29:
`which` is a shell command that will give the full path of the binary, so `which traffic-editor` will tell you the full path of the `traffic-editor` binary that is currently found using your `PATH` environment variable.

destkk, 2021-03-29:
> `which` is a shell command that will give the full path of the binary, so `which traffic-editor` will tell you the full path of the `traffic-editor` binary that is currently found using your `PATH` environment variable.

Hi this is the path `/opt/ros/foxy/bin/traffic-editor`
I have wiped out the previous workspace and trying to install this new one provided by you. https://github.com/open-rmf/rmf

codebot, 2021-03-29:
Just trying to make sure we're on the same page, because a lot of things have been moving rapidly with the re-organization of all the source code trees -- is your source tree for all things RMF fresh? In other words, is the directory tree where you ran `vcs import ...` fresh as of today?

There are basically two strategies to go forward: (1) using the binary packages, and trying to fix whatever the issue is there, or (2) uninstalling those binary packages and building everything from-source.

I would recommend strategy (2) to take advantage of the latest (including new traffic-editor features), but it's up to you; I think either strategy can be made to work. To be certain that the from-source build will succeed, I would suggest uninstalling the binary packages:
```
sudo apt-get remove ros-foxy-ament-cmake-catch2 ros-foxy-building-gazebo-plugins ros-foxy-building-map-msgs ros-foxy-building-map-tools ros-foxy-rmf-cmake-uncrustify ros-foxy-rmf-dispenser-msgs ros-foxy-rmf-door-msgs ros-foxy-rmf-fleet-adapter ros-foxy-rmf-fleet-msgs ros-foxy-rmf-lift-msgs ros-foxy-rmf-task-msgs ros-foxy-rmf-traffic-msgs ros-foxy-rmf-traffic-ros2 ros-foxy-rmf-traffic ros-foxy-rmf-utils ros-foxy-traffic-editor
```
Then, remove the `build` and `install` directories that the `colcon build` step created, to fully clean the cache, get a new terminal, and try the `colcon build` step again.

destkk, 2021-03-29:
> Just trying to make sure we're on the same page, because a lot of things have been moving rapidly with the re-organization of all the source code trees -- is your source tree for all things RMF [...]

Yes the vcs import is from the link you provided. May I know how to clear cache?

marcoag, 2021-03-29:
Ok, to avoid troubles I would advice try not to mix RMF sources and binaries as @codebot suggests.

If you go for sources, it seems from here:

> ```
> 
> DOWNLOADING MODELS WITH COMMAND: ros2 run building_map_tools building_map_model_downloader /home/rosuser/rmf_ws/src/demonstrations/rmf_demos/rmf_demos_maps/maps/office/office.building.yaml
> BUILDING WORLDFILE WITH COMMAND: ros2 run building_map_tools building_map_generator gazebo /home/rosuer/rmf_ws/src/demonstrations/rmf_demos/rmf_demos_maps/maps/triple_H/triple_H.building.yaml /home/rosuer/rmf_ws/build/rmf_demos_maps/maps/triple_H/triple_H.world /home/rosuser/rmf_ws/build/rmf_demos_maps/maps/triple_H/models
> DOWNLOADING MODELS WITH COMMAND: ros2 run building_map_tools building_map_model_downloader /home/rosuer/rmf_ws/src/demonstrations/rmf_demos/rmf_demos_maps/maps/triple_H/triple_H.building.yaml
> Package 'building_map_tools' not found
> make[2]: *** [CMakeFiles/generate_clinic_nav_graphs.dir/build.make:61: maps/clinic/nav_graphs/phony] Error 1
> make[1]: *** [CMakeFiles/Makefile2:118: CMakeFiles/generate_clinic_nav_graphs.dir/all] Error 2
> make: *** [Makefile:141: all] Error 2
> ---
> Failed   <<< rmf_demos_maps [2.45s, exited with code 2]
> ```

That the source was outdated as the package name for `building_map_tools` has changed to `rmf_building_map_tools` and this has already been addressed in https://github.com/open-rmf/rmf_traffic_editor/pull/310.

My advice is to remove all related rmf packages and try building from source.

If you follow the binary installation (you won't get the latest traffic editor features though) you can just follow the instructions in: 

https://osrf.github.io/ros2multirobotbook/#installation-of-the-rmf-essentials

codebot, 2021-03-29:
The cache is stored in the `build` directory tree. Removing the `build` and `install` directories will fully "reset" the build tree, since all the build products will be gone.

destkk, 2021-03-29:
> Just trying to make sure we're on the same page, because a lot of things have been moving rapidly with the re-organization of all the source code trees -- is your source tree for all things RMF [...]

Yes i have followed this step to uninstall all of the binary packages and colcon build the rmf_ws again. I am still receiving the building_map_tools not found error..

destkk, 2021-03-29:
I realised when i colcon build it, it is still running `ros2 run building_map_tools ...` and not `ros2 run rmf_building_map_tools ...`

codebot, 2021-03-29:
If your code tree was created before this PR went in (3 hours ago), that would explain it:
https://github.com/open-rmf/rmf_traffic_editor/pull/310/files

You can either `git pull` inside your clone of `rmf_traffic_editor` (which should be in `~/rmf_ws/src/rmf_traffic_editor`) or you could just remove your entire `~/rmf_ws` tree and follow the from-source instructions again to start with a "very very latest" source tree

codebot, 2021-03-29:
related PR in-process here will also be needed, for the same reason. Hopefully will merge (very) soon: https://github.com/open-rmf/rmf_demos/pull/17/files

destkk, 2021-03-29:
> related PR in-process here will also be needed, for the same reason. Hopefully will merge (very) soon: https://github.com/open-rmf/rmf_demos/pull/17/files

Hi there. By manually editing all the required files to `rmf_building_map_tools `it could colcon build perfectly.
The path for the traffic editor is this:` /home/rosuser/rmf_ws/install/rmf_traffic_editor/bin/traffic-editor`

I am still unable to see the image when I open `office.building.yaml` after sourcing `setup.bash`

[image]

codebot, 2021-03-29:
Thanks for the patience, and I'm glad to hear that you were able to build the very latest `traffic-editor` from source. It's nice to see that `Building` menu in the upper-left :smile: 

Where are you loading the `office.building.yaml` file from? Can you copy-paste the first ~20 or so lines of the console output of `traffic-editor` here?

destkk, 2021-03-29:
> Thanks for the patience, and I'm glad to hear that you were able to build the very latest `traffic-editor` from source. It's nice to see that `Building` menu in the upper-left smile
>
> Where are you [...]

Thank you too for your patience in guiding me. 
May I know what is the error below too?

```
traffic-editor 
settings filename: [/home/rosuser/.config/open-robotics/traffic-editor.conf]
Initialize crowd_sim_implementation for building
couldn't parse /home/rosuser/rmf_demos_ws/install/traffic_editor_assets/share/assets/thumbnails/model_list.yaml: bad file
Building::load(/home/rosuser/rmf_ws/install/rmf_demos_maps/share/rmf_demos_maps/office/office.building.yaml)
changing directory to [/home/rosuser/rmf_ws/install/rmf_demos_maps/share/rmf_demos_maps/office]
parsing level [L1]
  level L1 drawing: office.png
used 3 measurements to estimate meters/pixel as 0.00847
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlack
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlack
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlack
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlack
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlack
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlack
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlack
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlack
[ERROR] No thumbnail found: OpenRobotics/Fridge
[ERROR] No thumbnail found: OpenRobotics/OfficeChairGrey
[ERROR] No thumbnail found: OpenRobotics/AdjTable
[ERROR] No thumbnail found: OpenRobotics/AdjTable
[ERROR] No thumbnail found: OpenRobotics/AdjTable
[ERROR] No thumbnail found: OpenRobotics/SmallCubicle
[ERROR] No thumbnail found: OpenRobotics/SmallCubicle
[ERROR] No thumbnail found: OpenRobotics/SmallCubicle
[ERROR] No thumbnail found: OpenRobotics/SmallCubicle
[ERROR] No thumbnail found: OpenRobotics/SmallCubicle
[ERROR] No thumbnail found: OpenRobotics/SmallCubicle
[ERROR] No thumbnail found: OpenRobotics/SmallCubicle
[ERROR] No thumbnail found: OpenRobotics/SmallCubicle
[ERROR] No thumbnail found: OpenRobotics/AdjTable
[ERROR] No thumbnail found: OpenRobotics/Drawer
[ERROR] No thumbnail found: OpenRobotics/BigCubicle
[ERROR] No thumbnail found: OpenRobotics/BigCubicle
[ERROR] No thumbnail found: OpenRobotics/BigCubicle
[ERROR] No thumbnail found: OpenRobotics/BigCubicle
[ERROR] No thumbnail found: OpenRobotics/AdjTable
[ERROR] No thumbnail found: OpenRobotics/AdjTable
[ERROR] No thumbnail found: OpenRobotics/AdjTable
[ERROR] No thumbnail found: OpenRobotics/AdjTable
[ERROR] No thumbnail found: OpenRobotics/AdjTable
[ERROR] No thumbnail found: OpenRobotics/AdjTable
[ERROR] No thumbnail found: OpenRobotics/OfficeChairGrey
[ERROR] No thumbnail found: OpenRobotics/OfficeChairGrey
[ERROR] No thumbnail found: OpenRobotics/OfficeChairGrey
[ERROR] No thumbnail found: OpenRobotics/OfficeChairGrey
[ERROR] No thumbnail found: OpenRobotics/OfficeChairGrey
[ERROR] No thumbnail found: OpenRobotics/OfficeChairGrey
[ERROR] No thumbnail found: OpenRobotics/OfficeChairGrey
[ERROR] No thumbnail found: OpenRobotics/OfficeChairGrey
[ERROR] No thumbnail found: OpenRobotics/OfficeChairGrey
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlue
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlue
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlue
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlue
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlue
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlue
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlue
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlue
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlue
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlue
[ERROR] No thumbnail found: OpenRobotics/OfficeChairBlack
[ERROR] No thumbnail found: OpenRobotics/OfficeChairGrey
[ERROR] No thumbnail found: OpenRobotics/Drawer
[ERROR] No thumbnail found: OpenRobotics/Drawer
[ERROR] No thumbnail found: OpenRobotics/ConfTable
[ERROR] No thumbnail found: OpenRobotics/RecTable
[ERROR] No thumbnail found: OpenRobotics/WhiteCabinet
[ERROR] No thumbnail found: OpenRobotics/SquareShelf
[ERROR] No thumbnail found: OpenRobotics/TeleportDispenser
[ERROR] No thumbnail found: OpenRobotics/Coke
[ERROR] No thumbnail found: OpenRobotics/TeleportIngestor
restoring viewport: (0.0, 0.0, 0.000000)
```

codebot, 2021-03-29:
Ah! So the core of the problem is that last line: the last number is the scale factor, which is set to `0.000000` which means that it's infinitely zoomed-out. This is a silly bug. I'll fix it in a separate PR to make sure that the zoom level doesn't get stuck to zero :facepalm:  the fastest way for you to resolve this is to edit the QSettings file that is (not helpfully) restoring the zoom level to `0` automatically on startup:
```
~/.config/open-robotics/traffic-editor.conf
```
The underlying problem is that it's really easy to set an extremely small zoom level by using the mouse scroll wheel when there isn't a level image shown. Then, when a level image actually is loaded, it's smaller than one pixel.

I'll fix that now. But to become un-stuck in the meantime, you can edit the `viewport_scale` line in your QSettings file (at the path above) to `1.0` or just remove that file and it will auto-generate next time you start `traffic-editor`.

destkk, 2021-03-29:
It is working now. I have temporary edited the viewport_scale to 1.0 and it works fine for now. Thank you so much for your patience in resolving this issue for me!

codebot, 2021-03-29:
Yes, the thumbnail directory needs to be set to your new installation. If you go to `Edit...Preferences...` there will be a box in there where you can navigate to the thumbnail directory. You'll want something like this:
```
/home/rosuser/rmf_ws/src/rmf_traffic_editor/rmf_traffic_editor_assets/assets/thumbnails
```

destkk, 2021-03-29:
> Yes, the thumbnail directory needs to be set to your new installation. If you go to `Edit...Preferences...` there will be a box in there where you can navigate to the thumbnail directory. You'll [...]

Thank you. This works perfectly.

destkk, 2021-03-29:
Sorry, just another error i faced. After i have created a new building.yaml file and saved, traffic-editor closes and the CLI shows this error 

```
Building::save_yaml()
Segmentation fault (core dumped)

```

codebot, 2021-03-29:
mmm. that sounds like a new problem. Can you create a new ticket for that? I'll try to re-create it also.

codebot, 2021-03-29:
It might be what I just fixed in #312 as I was testing this today, can you try to `git pull` your clone of `rmf_traffic_editor`, build again, and see if it's fixed?

fastest way to rebuild just the traffic editor:
```
cd ~/rmf_ws
colcon build --packages-select rmf_traffic_editor
```

destkk, 2021-03-29:
Sorry may I know how to git pull?

codebot, 2021-03-29:
to update your copy of the `rmf_traffic_editor` repository:
```
cd ~/rmf_ws/src/rmf_traffic_editor
git pull
```
then rebuild traffic-editor:
```
cd ~/rmf_ws
colcon build --packages-select rmf_traffic_editor
```

destkk, 2021-03-29:
> to update your copy of the `rmf_traffic_editor` repository:
> 
> ```
> cd ~/rmf_ws/src/rmf_traffic_editor
> git pull
> ```
> 
> then rebuild traffic-editor:
> 
> ```
> cd ~/rmf_ws
> colcon build --packages-select rmf_traffic_editor
> ```

Ok. It works perfectly after git pull. But this issue may still come back again I think that it would be good if we could find out the source of this error. Thank you so much.

codebot, 2021-03-29:
I believe it was fixed in #312 which you have now incorporated into your local build. But if you see any crashes please do create tickets for them. I'll close the other ticket now.

codebot, 2021-03-29:
closing this issue now, but feel free to re-open or create a new ticket if something bad occurs again.

closed 2021-03-29 by codebot