open-rmf/rmf_traffic_editor #317: Segmentation fault (core dumped) error when creating new building.yaml file

destkk, 2021-03-29:
Ubuntu 20.04

Hi there. I have faced a segmentation fault (core dumped) error after creating a new building.yaml file and saved, traffic editor closes and the CLI shows the following error,

It was working fine previously. I realized this error happens after I have created fiducials and deleted the building.yaml file and recreating a new one.

Building::save_yaml()
Segmentation fault (core dumped)
settings filename: [/home/rosuser/.config/open-robotics/traffic-editor.conf]
Initialize crowd_sim_implementation for building
parsed /home/rosuser/rmf_ws/src/traffic_editor/rmf_traffic_editor/rmf_traffic_editor_assets/assets/thumbnails/model_list.yaml successfully
restoring viewport: (0.0, 0.0, 0.450514)
set building filename to [/home/rosuser/rmf_ws/install/rmf_demos_maps/share/rmf_demos_maps/testingmap1/testingmap1.building.yaml]
nothing to draw!
Building::save_yaml(/home/rosuser/rmf_ws/install/rmf_demos_maps/share/rmf_demos_maps/testingmap1/testingmap1.building.yaml)
Segmentation fault (core dumped)

codebot, 2021-03-29:
Thanks for creating this ticket. If you `git pull` your source tree in `rmf_traffic_editor` so that it includes #312  , does that resolve this issue?

destkk, 2021-03-29:
Hi there. May I know how do I git pull this?

codebot, 2021-03-29:
fixed by #312

closed 2021-03-29 by codebot