open-rmf/rmf_traffic_editor #319: Unable to generate world file and models from 3 levels map

destkk, 2021-03-30:
I'm trying to generate a .world file and models from the 3 levels map that I have built in traffic editor using this command 
`ros2 run rmf_building_map_tools building_map_generator gazebo  ~/rmf_ws/install/rmf_demos_maps/share/rmf_demos_maps/test_building.building.yaml test_building.world ~/rmf_ws/install/rmf_demos_maps/share/rmf_demos_maps/maps/test_building/models` but I am receiving this error and unable to build the map. Could it be possible that I have built the map wrongly?

```
building name: test_building
parsing level L1
parsing level L2
parsing level L3
calculating levels relative to L1
level L1 scale: 0.049999520301876176
calculating level L2 offset and scale...
  4 common fiducials:
Transform.set_from_fiducials()
  4 fiducial pairs:
    (7814.1, -5614.0) -> (7830.4, -5606.0)   ref1
    (7842.3, -5621.7) -> (7860.4, -5614.8)   ref2
    (7835.4, -5535.4) -> (7852.4, -5521.5)   ref3
    (7861.7, -5543.8) -> (7881.6, -5530.4)   ref4
    calc dist ref1 <=> ref2
    calc dist ref1 <=> ref3
    calc dist ref1 <=> ref4
    calc dist ref2 <=> ref3
    calc dist ref2 <=> ref4
    calc dist ref3 <=> ref4
Bearings:
[(-0.2673258533057868, -0.2847918192598588), (1.3065081154511626, 1.316608516250293), (0.9749057912340271, 0.9755285900765562), (1.6511376819099384, 1.6569610474329988), (1.3266570329172527, 1.324780805386846), (-0.30583893247514277, -0.29509117110947225)]
  -0.017465965954072016
  0.010100400799130327
  0.0006227988425291375
  0.0058233655230603976
  -0.0018762275304067266
  0.010747761365670516
  Circular mean: -0.0013254962167178644
Distances:
[(29.24060498690073, 31.27161556427871), (81.37131959111895, 87.30634064602675), (84.84514390935956, 91.31660153553688), (86.54917570953565, 93.63437260429552), (80.3280698448556, 87.01600818240259), (27.645917329689453, 30.5714307973966)]
mean relative scale: 1.0816019487363226
translation: (29.086, -21.052)
calculating level L3 offset and scale...
  4 common fiducials:
Transform.set_from_fiducials()
  4 fiducial pairs:
    (7814.1, -5614.0) -> (7828.2, -5602.9)   ref1
    (7842.3, -5621.7) -> (7857.9, -5612.4)   ref2
    (7835.4, -5535.4) -> (7850.0, -5521.2)   ref3
    (7861.7, -5543.8) -> (7878.9, -5529.9)   ref4
    calc dist ref1 <=> ref2
    calc dist ref1 <=> ref3
    calc dist ref1 <=> ref4
    calc dist ref2 <=> ref3
    calc dist ref2 <=> ref4
    calc dist ref3 <=> ref4
Bearings:
[(-0.2673258533057868, -0.3113945432709235), (1.3065081154511626, 1.310541425027987), (0.9749057912340271, 0.9638479321628333), (1.6511376819099384, 1.6571453588867642), (1.3266570329172527, 1.321115071181614), (-0.30583893247514277, -0.2908862733686624)]
  -0.04406868996513669
  0.004033309576824351
  -0.011057859071193832
  0.006007676976825804
  -0.005541961735638701
  0.01495265910648036
  Circular mean: 0.005944596558665005
Distances:
[(29.24060498690073, 31.146942867639737), (81.37131959111895, 84.5306178848827), (84.84514390935956, 88.89417918514198), (86.54917570953565, 91.5681611587782), (80.3280698448556, 85.20200941879283), (27.645917329689453, 30.219519999496686)]
mean relative scale: 1.0605833738230235
translation: (20.109, -18.711)
parsing lift L1_lift
generating model of level L1 in /home/rosuser/rmf_ws/install/rmf_demos_maps/share/rmf_demos_maps/maps/test_building/models/test_building_L1
  wrote /home/rosuser/rmf_ws/install/rmf_demos_maps/share/rmf_demos_maps/maps/test_building/models/test_building_L1/model.config
generating floor polygon 1 on floor test_building_L1
hole vertices: []
lift vertices: {'L1_lift': [(393.88725157843004, -273.61961071600194), (391.19645784727487, -284.6151565913269), (399.20513944992985, -286.5750147462187), (401.895933181085, -275.5794688708937)]}
Traceback (most recent call last):
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/rmf_building_map_tools/building_map_generator", line 11, in <module>
    load_entry_point('rmf-building-map-tools==0.0.0', 'console_scripts', 'building_map_generator')()
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map_generator/building_map_generator.py", line 16, in main
    g.generate_gazebo_sdf(
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map/generator.py", line 59, in generate_gazebo_sdf
    self.generate_sdf(
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map/generator.py", line 41, in generate_sdf
    building.generate_sdf_models(output_models_dir)
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map/building.py", line 233, in generate_sdf_models
    level.generate_sdf_model(model_name, model_path)
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map/level.py", line 273, in generate_sdf_model
    self.write_sdf(model_name, model_path)
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map/level.py", line 258, in write_sdf
    self.generate_floors(model_ele, model_name, model_path)
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map/level.py", line 241, in generate_floors
    floor.generate(
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map/floor.py", line 231, in generate
    self.triangulate_polygon(self.polygon, triangles)
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map/floor.py", line 131, in triangulate_polygon
    self.triangulate_polygon(item, triangles)
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map/floor.py", line 118, in triangulate_polygon
    self.triangulate_polygon(poly, triangles)
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map/floor.py", line 118, in triangulate_polygon
    self.triangulate_polygon(poly, triangles)
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map/floor.py", line 118, in triangulate_polygon
    self.triangulate_polygon(poly, triangles)
  [Previous line repeated 977 more times]
  File "/home/rosuser/rmf_ws/install/rmf_building_map_tools/lib/python3.8/site-packages/building_map/floor.py", line 102, in triangulate_polygon
    poly = shapely.geometry.polygon.orient(geom)
  File "/usr/lib/python3/dist-packages/shapely/geometry/polygon.py", line 396, in orient
    if signed_area(ring)/s >= 0.0:
  File "/usr/lib/python3/dist-packages/shapely/algorithms/cga.py", line 6, in signed_area
    xs, ys = ring.coords.xy
  File "/usr/lib/python3/dist-packages/shapely/geometry/base.py", line 328, in _get_coords
    if self.is_empty:
  File "/usr/lib/python3/dist-packages/shapely/geometry/base.py", line 672, in is_empty
    return (self._geom is None) or bool(self.impl['is_empty'](self))
  File "/usr/lib/python3/dist-packages/shapely/predicates.py", line 25, in __call__
    return self.fn(this._geom)
  File "/usr/lib/python3/dist-packages/shapely/geos.py", line 579, in errcheck_predicate
    if result == 2:
RecursionError: maximum recursion depth exceeded in comparison

```

codebot, 2021-03-30:
Interesting, I haven't seen that error from `shapely` before. Have you defined polygon(s) for your floors? If so, do they have a (very) unusual shape? I'd be happy to look at the `.building.yaml` file if you could send it to me.

destkk, 2021-03-30:
Hi there. Do I send it to you via email?

destkk, 2021-03-30:
> Interesting, I haven't seen that error from `shapely` before. Have you defined polygon(s) for your floors? If so, do they have a (very) unusual shape? I'd be happy to look at the `.building.yaml` file if you could send it to me.

I have just emailed you the building file!

codebot, 2021-03-30:
I believe this is fixed on #321 but we are going to do a bit more testing first before merging, since there is a bit of magic geometric algorithm involved.

codebot, 2021-03-30:
#321 is now merged. Thanks @Yadunund for the testing and extra fixes!

closed 2021-03-30 by codebot