open-rmf/rmf_traffic_editor #371: [feature request] collinear vertices

Yadunund, 2021-07-27:
It is desirable to select 3 vertices on the map and "collinearize" them such that the 3rd vertex get projected onto the line passing through the first two. This is very useful when drawing traffic lanes as it will allow the RMF traffic planner to generate routes such that the 2nd (middle) vertex is skipped if the robot is travelling from 1st -> 3rd vertices. But having this 2nd vertex is helpful during negotiations as the robot can pull into this spot if necessary.

codebot, 2021-07-28:
Working on this now, should be able to hammer something out quickly today. The main "UI change" will be to allow multi-select of the vertices that need to move to become colinear. I think something like shift-click to "add to selection" should work fine, and then a menu command (for now at least) to "make colinear."

codebot, 2021-08-30:
implemented in #372

closed 2021-08-30 by codebot