open-rmf/rmf_traffic_editor #381: Feature Request: Be able to add world plugins to map

mxgrey, 2021-08-16:
It's convenient to have build pipelines that automatically generate SDF world files from a traffic editor map, but it looks like there is not currently a way to include a world plugin in the SDF world file that gets generated by traffic-editor. It would be nice to add that functionality, because without it, we need to manually edit the generated world files to add a world plugin.

codebot, 2021-08-23:
How should this work from a UI perspective? I would guess that people would prefer to use their own code editor to write the SDF XML (rather than just a multi-line textbox in Traffic Editor). "What if" there was a top-level parameter that was just a relative filename of an XML snippet to insert into the generated world SDF?

mxgrey, 2021-08-24:
Sure, that sounds like a good approach to me :+1: