open-rmf/rmf_traffic_editor #386: Invalid cross-device link for certain fuel models

cnboonhan, 2021-09-06:
I encountered some issues with running airport_terminal under ignition gazebo, where some models ( such as Construction Cone ) could not be found ( and this prevented ignition gazebo from fully launching ) , and so I tried to debug by running this command standalone:

```
 ros2 run rmf_building_map_tools building_map_model_downloader install/rmf_demos_maps/share/rmf_demos_maps/airport_terminal/airport_terminal.building.yaml -f -e install/rmf_demos_maps/share/rmf_demos_maps/airport_terminal/`
```

For certain models, the following error appears:
```
ERROR::pit_crew.download_model_fuel_tools():702: Could not download model 'Construction Cone'! [Errno 18] Invalid cross-device link: '/root/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/construction%20cone' -> '/root/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/construction cone'
```

codebot, 2021-09-06:
Is it all the models that have spaces in their names?

SpacesShouldBeBannedSinceTheyAreUnnecessary

or_we_can_use_underscores_maybe_question_mark

cnboonhan, 2021-09-06:
you are right, only those with whitespaces explode. I didn't notice that before. However, additionally, this only happens when the corresponding fuel model exists in `~/.ignition/fuel/models`. Meaning to say, if i delete that folder, and run `rmf_building_map_downloader` for `airport_terminal_ign`, everything works. 

Similarly, if i delete that folder, and run `rmf_building_map_downloader` for `airport_terminal` (classic gazebo), it works.

However, if I run `building_map_model_downloader` for `airport_terminal` (gz) and then `airport_terminal_ign`, then only those models with whitespaces break. :exploding_head:

aaronchongth, 2021-09-07:
I think a good starting point would be revisiting the effort for downloading using ignition fuel tools
https://github.com/open-rmf/rmf_traffic_editor/pull/359

and the subsequent bug fixes after the migration, here are some of them
https://github.com/open-rmf/rmf_traffic_editor/issues/360
https://github.com/open-rmf/rmf_traffic_editor/issues/361

Another big change to all the model downloading behavior would be this, https://github.com/open-rmf/rmf_demos/pull/68.

aaronchongth, 2021-09-07:
I find it curious that `Lamp Post` and `Standing Person` managed to make their way into `install/rmf_demos_maps/share/rmf_demos_maps/maps/airport_terminal/models/` but not `install/rmf_demos_maps/share/rmf_demos_maps/maps/airport_terminal_ign/models/` :thinking:

cnboonhan, 2021-09-07:
reckon thats where whitespace in the name caused some explosions :)

aaronchongth, 2021-09-07:
Just did a simple build up to `rmf_demos_maps`, with clean model directories everywhere, here are the results,

Downloaded from Fuel, note the duplicated models with spaces, for example, `'cafe table'` and `cafe%20table`, also note that `'lamp pose'` and `'standing person'` did not get duplicated,
```
aaron@excellence2:~/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models$ ls
 adjtable              electronicsrecycling    mopcart2            shower
 ambulance             femalevisitor           mopcart3            smallcubicle
 ambulance2            femalevisitorsit        normalbed           sofa
 bathroomsink          fountain                nurse               squareshelf
 bigcubicle            fridge                  nursedesk          'standing person'
 bookshelf             gazebo                  nursetable          storagerack
 cafe                  ivstand                 oak%20tree          storagerackcovered
 cafe%20table          kitchencountertop      'oak tree'           storagerackcoveropen
'cafe table'           kitchensink             officechairblack    surgicaltrolleymed
 cgmvanguard           lampandstand            officechairblue     suv
 chair                'lamp post'              officechairgrey     table
 coffeetable           litterbin               opscrubs            toilet
 coke                  maintable               parkingtrolleymax   trashbin
 conftable             malevisitoronphone      patientfsit         trolleybox1
 construction%20cone   malevisitorphone        patientwheelchair   truckbox
'construction cone'    malevisitorsit          pine%20tree         truckdelivery
 desk                  manualfirealarmbutton  'pine tree'          tvstand
 deskchair             metalcabinet            playground          vendingmachine
 drawer                minisofa                potatochipchair     visitorkidsit
 elderladypatient      monitorandkeyboard      rectable            whitecabinet
 eldermalepatient      mopcart1                scrubs              whitechipchair
```

Install forlder for `rmf_demos_maps` for `*_ign` maps,
```
aaron@excellence2:~/workspaces/map_tools/install/rmf_demos_maps/share/rmf_demos_maps/maps/airport_terminal_ign/models$ ls
airport_terminal_L1
```

Install folder for `rmf_demos_maps` for non-`*_ign` maps, note that `'Lamp Post` and `Standing person` got copied over but not the rest of the models with spaces in their names,
```
aaron@excellence2:~/workspaces/map_tools/install/r
mf_demos_maps/share/rmf_demos_maps/maps/airport_temf_demos_maps/share/rmf_demos_maps/maps/airport_terrminal/models$ ls
 airport_terminal_L1    FemaleVisitorSit    LampAndStand       MopCart3            SUV
 Bookshelf              Fountain           'Lamp Post'         ParkingTrolleyMax   Table
 Cafe                   Fridge              MainTable          Playground          VisitorKidSit
 Chair                  Gazebo              MaleVisitorPhone   PotatoChipChair     WhiteChipChair
 ElectronicsRecycling   KitchenCountertop   MaleVisitorSit    'Standing person'
 FemaleVisitor          KitchenSink         MiniSofa           StorageRack
```

Issues:
* Duplicated models that have spaces in their names polluting the ignition fuel model directory
* No models are exported to the map folders for ignition
* Models with spaces in their names, the ones that got duplicated are not exported to the non-ignition map folder, while the ones that did not get duplicated, for example `Lamp Post` and `Standing person` managed to get exported.

cnboonhan, 2021-09-08:
thanks for the detailed data! Happy to look into this if no one else already is, i'm helping to patch up related issue on `test_ackman`

luca-della-vedova, 2021-09-08:
>     * No models are exported to the map folders for ignition

I can see why this decision was made, otherwise there would be duplicated models in the binaries and they would be double the size. The Ignition launch file [looks for the models](https://github.com/open-rmf/rmf_demos/blob/main/rmf_demos_ign/launch/simulation.launch.xml#L10) also in the Gazebo folder.
Now this creates a bit of a conflict since for example for a level there will be multiple models with the same model (i.e. `building_L1`) and it is unclear which one will be loaded, however they happen to be the same so this is actually a non issue.
A real fix would be to throw all the models, including the building and floor generated by `rmf_building_map_tools` into a common folder, to also remove the current mesh duplication.

aaronchongth, 2021-09-08:
Fair enough, I like the idea of merging the model folders into a common one. We can also take this opportunity to deprecate the use of downloading models using `http` requests, and any mention of the `~/.gazebo` folder

cnboonhan, 2021-09-08:
should we do an immediate patch https://github.com/open-rmf/rmf_traffic_editor/pull/387 so the model exports work, and then do a more complex refactor?

aaronchongth, 2021-09-08:
I would think a patch before a refactor makes more sense. Let's keep the changes small

aaronchongth, 2021-09-10:
Now that https://github.com/open-rmf/rmf_traffic_editor/pull/387 has been merged, I'll close this issue to open another which details refactoring plans.

closed 2021-09-10 by aaronchongth