open-rmf/rmf_traffic_editor #388: Deprecate option to use http request for model downloading

aaronchongth, 2021-09-10:
### Deprecation

Related to https://github.com/open-rmf/rmf_traffic_editor/issues/386.

We should use `ign-fuel-tools` moving forwards, and support upstream development if we need any new features.

cnboonhan, 2021-09-13:
happy to work on this, but I would like to get a clear idea of the ideal result. From [here](https://github.com/open-rmf/rmf_traffic_editor/issues/386#issuecomment-914999932): 
```
A real fix would be to throw all the models, including the building and floor generated by rmf_building_map_tools into a common folder, to also remove the current mesh duplication.
```

I am guessing, we the model download process should be something like
* download from fuel to `~/.ignition`, ignoring duplicates
* export models to dedicated `install` folder, perhaps `install/rmf_fuel_models`
* Link all launch files to this folder 

Some thoughts I had:
* Ignition seems to halt if some models are missing. This might not be ideal for CI, which might want to omit some models for the sake of efficiency. Is there a way to let ignition continue running with some missing models? 
* Would there be value in improving the old models we have to remove whitespace, etc? @aaronchongth  mentioned some stuff about some standards to enforce on fuel models, which we could spend some time to tidy up if ideal.

aaronchongth, 2021-09-13:
As per VC, this issue will only be related to refactoring and deprecating the option to download models using the `http` request feature.

Regarding, the discussion about exporting models, I have created a separate issue https://github.com/open-rmf/rmf_demos/issues/89.

As for reducing the number of workarounds we apply during model downloading, e.g. handling white space, changing names, etc, as you surmised, the ideal solution would be to improve the older models on Fuel, however I would expect plenty of users are using them as it is, and any major changes would break lots of simulation behaviors across the board. We should definitely revisit this once we get a better understanding on our simulation model needs some time soon. Thanks!

cnboonhan, 2021-09-13:
yes thanks for clarifying! will dive in 🤽