open-rmf/rmf_traffic_editor #413: Feature Request: Be able to draw cleaning paths

mxgrey, 2022-01-03:
At the moment the fleet adapters are using a somewhat roundabout strategy to inform the fleet adapter about where and how robots can perform cleaning tasks. It would be nice if we could incorporate that information directly into the nav graph yaml file, drawn and exported using the traffic editor.

I imagine we'd want to add these features:
* Create a new type of nav graph entries parallel to `lanes` and `vertices`, maybe call it `paths`
* Give each `paths` element a sequence of vertices that don't (necessarily) exist on the nav graph
* Give the `paths` elements attributes/tags, similar to what we do for vertices

Maybe this could be a format to consider:
```
paths:
- name: my_clean_path
  enter_from: [3, 7, ... ]
  path: [ [10.12, -17.43], [11.12, -17.43], ... ]
  attributes:
    clean:
      zone: zone_1
      types: ['vacuum', 'mop']
```

mxgrey, 2022-01-03:
@Yadunund might be interested in this topic.

Yadunund, 2022-01-03:
This would be very useful 😃