open-rmf/rmf_traffic_editor #416: Lift falling problem

ClobotLucy, 2022-01-12:
I added Lift to allow movement between two levels via traffic-editor.
I created a Gazebo simulation file with Building_map_tools.
But when I run the simulation, the lift falls.

[image]

How can I solve this?
Thanks.

youliangtan, 2022-01-12:
I think the issue is related to the missing rmf lift plugin for the lift. This plugin is part of [rmf_simulation](https://github.com/open-rmf/rmf_simulation), and it handles the logic for rmf to interact with the lift model.

When you look into the generated world file, the lift plugin is indicated as followed:
```xml
      <plugin name="lift" filename="liblift.so">
        <lift_name>Lift2</lift_name>
        <floor name="L1" elevation="0.0">
         ...
      </plugin>
```
I tried to remove the above block in my testing world, and I am able to recreate the free fall lift. So, try to have `rmf_simulation` lift plugin availbele and check if it works.

ClobotLucy, 2022-01-13:
Thanks for your reply.
I ran gazebo11 on ros noetic.

1. The liblift.so and libdoor.so files are created through the rmf_building_sim_gazebo_plugins package, so are they not usable in noetic?

2. If I run the gazebo simulation in ros2, can I check the topics related to the door or lift?

Thank you.

youliangtan, 2022-01-15:
If you check the dependency [here](https://github.com/open-rmf/rmf_simulation/blob/5d0f65a4761a8fff7c775b2d3e5d26a4fb99c10b/rmf_building_sim_gazebo_plugins/CMakeLists.txt#L23-L24), `rmf_building_sim_gazebo_plugins` requires ros2. Unfortunately, it is not supported in ros noetic, but surely you can easily create your own custom plugin, then swap it in the `xml world` file.

To check the topics related to doors and lifts, you can just `ros2 topic list`. The plugins listen to `/door_requests` and `/lift_requests`, publish `/door_states` and `/lift_states` respectively.

ClobotLucy, 2022-01-17:
When I run Gazebo (ros2), I got the following error: (Lift not included)
[image]

Any ideas what the problem is?

( Image drawn through traffic editor )
[image]

( Gazebo world file created through building_map_tools command )
```
ros2 run building_map_tools building_map_generator gazebo suntech_office.building.yaml suntech_office.world .
```

```
<sdf version="1.7">
  <world name="world">
    <scene>
      <ambient>0.8 0.8 0.8 1.0</ambient>
      <background>0 0 0</background>
      <grid>false</grid>
    </scene>
    <include>
      <uri>model://sun</uri>
    </include>
    <gui>
      <camera name="gui_camera">
        <pose>5.668941591994157 -13.900303438941659 10 0 0.6 1.57</pose>
      </camera>
      <plugin name="toggle_charging" filename="libtoggle_charging.so" />
      <plugin name="toggle_floors" filename="libtoggle_floors.so">
        <floor name="L1" model_name="suntech_office_L1">
          <model name="OpenRobotics/OfficeChairGrey" />
          <model name="OpenRobotics/OfficeChairGrey_2" />
          <model name="OpenRobotics/OfficeChairGrey_3" />
          <model name="OpenRobotics/AdjTable" />
          <model name="OpenRobotics/AdjTable_2" />
          <model name="OpenRobotics/AdjTable_3" />
          <model name="OpenRobotics/AdjTable_4" />
          <model name="OpenRobotics/AdjTable_5" />
          <model name="OpenRobotics/AdjTable_6" />
          <model name="OpenRobotics/AdjTable_7" />
          <model name="OpenRobotics/AdjTable_8" />
          <model name="OpenRobotics/AdjTable_9" />
          <model name="OpenRobotics/AdjTable_10" />
          <model name="OpenRobotics/AdjTable_11" />
          <model name="OpenRobotics/AdjTable_12" />
          <model name="OpenRobotics/AdjTable_13" />
          <model name="OpenRobotics/AdjTable_14" />
          <model name="OpenRobotics/AdjTable_15" />
          <model name="OpenRobotics/OfficeChairGrey_4" />
          <model name="OpenRobotics/OfficeChairGrey_5" />
          <model name="OpenRobotics/OfficeChairGrey_6" />
          <model name="OpenRobotics/OfficeChairGrey_7" />
          <model name="OpenRobotics/OfficeChairGrey_8" />
          <model name="OpenRobotics/OfficeChairGrey_9" />
          <model name="OpenRobotics/OfficeChairGrey_10" />
          <model name="OpenRobotics/OfficeChairGrey_11" />
          <model name="OpenRobotics/OfficeChairGrey_12" />
          <model name="OpenRobotics/OfficeChairGrey_13" />
          <model name="OpenRobotics/OfficeChairGrey_14" />
          <model name="OpenRobotics/OfficeChairGrey_15" />
          <model name="OpenRobotics/OfficeChairBlue" />
          <model name="OpenRobotics/OfficeChairBlue_2" />
          <model name="OpenRobotics/OfficeChairBlue_3" />
          <model name="OpenRobotics/OfficeChairBlue_4" />
          <model name="OpenRobotics/Table" />
          <model name="OpenRobotics/Table_2" />
          <model name="OpenRobotics/DeskChair" />
          <model name="OpenRobotics/DeskChair_2" />
          <model name="OpenRobotics/SquareShelf" />
[... 245 lines cut ...]
    <include>
      <name>OpenRobotics/SquareShelf_2</name>
      <uri>model://SquareShelf</uri>
      <pose>15.385471119219417 2.643623461629735 0.0 0 0 -3.1416</pose>
      <static>True</static>
    </include>
    <include>
      <name>OpenRobotics/SquareShelf_3</name>
      <uri>model://SquareShelf</uri>
      <pose>14.527883993418785 2.6528269828601783 0.0 0 0 -3.1416</pose>
      <static>True</static>
    </include>
    <include>
      <name>suntech_office_L1</name>
      <uri>model://suntech_office_L1</uri>
      <pose>0 0 0.0 0 0 0</pose>
    </include>
    <rmf_charger_waypoints name="charger_waypoints">
      <rmf_vertex name="n2" x="-0.20640027939747807" y="2.5332314993301583" level="L1" />
      <rmf_vertex name="n1" x="-0.10299896983577853" y="0.9960422839832" level="L1" />
      <rmf_vertex name="n19" x="13.770077119646563" y="10.596471654047377" level="L1" />
    </rmf_charger_waypoints>
  </world>
</sdf>
```

ClobotLucy, 2022-01-18:
Hello,

I reinstalled the rmf packages and ran it.
In Gazebo, the lift is also in normal condition.

Perhaps there was a problem with the old workspace.
Thanks.
[image]

youliangtan, 2022-01-18:
Nice then, I will consider this as solved. :)

closed 2022-01-18 by youliangtan