open-rmf/rmf_traffic_editor #417: Description of generated *.mlt and *.obj files

ljaniec, 2022-01-14:
## Feature request

Add detailed file format specifications (`meshes/*.obj`, `meshes/*.mtl` and `meshes/model.material` files + `model.config` and `model.sdf`) to support external automatic generation of walls etc.

### Description

I would like to see more documentation about the specifications of said files so that our developers can partially automate the creation of the simulation world.

Right now I didn't see description of these formats, e.g. for the simplest world with one 10mx10m floor and a single wall of length x height 5m x 1m with a measurement edge of length 1m for scale + one OpenRobotics model of scrubber I got these files:
[image]

The whole output is in the attachments.

### Implementation considerations

Detailed description with maybe example Python script generation for walls would be enough, I think.

This format seems to be simple enough to use (unlike Collada files *.dae) with scripts, so it would be easier for automated generation of example worlds with mazes and so on.

[example_world_output.zip](https://github.com/open-rmf/rmf_traffic_editor/files/7870442/example_world_output.zip)

luca-della-vedova, 2022-01-24:
Hi @ljaniec!

The walls / floors are generated based on the edges (walls) and polygons (floors) defined in `traffic-editor`.
The generation itself is done by `rmf_building_map_tools`, specifically by the [generate](https://github.com/open-rmf/rmf_traffic_editor/blob/main/rmf_building_map_tools/building_map/floor.py#L154) [functions ](https://github.com/open-rmf/rmf_traffic_editor/blob/main/rmf_building_map_tools/building_map/wall.py#L222).
These functions create both the obj, and mtl for the collision / visual as well as the xml tags in the `sdf` file.
The creation of the `obj` files is somewhat complex since it involves generating 3d meshes starting from 2d parameters, and that involves generating the polygons (triangles) and mapping the textures so they look as intended.

Generally speaking, we group autogenerated meshes, like walls, into single obj / mtl file sets that share the same property (i.e. texture, transparency) as you can see [here](https://github.com/open-rmf/rmf_traffic_editor/blob/main/rmf_building_map_tools/building_map/level.py#L221-L229). The reason behind this is that for performance reasons we would want to have the least number of models possible but still have the flexibility of being able to have arbitrary parameters for each wall section. In your map you have one floor and one wall which makes me think that all the walls have the same textures / parameters, as well as all the floor polygons.

Models themselves (i.e. you mentioned scrubber) are downloaded separately at build time. The [building_map_model_downloader](https://github.com/open-rmf/rmf_traffic_editor/tree/main/rmf_building_map_tools/building_map_model_downloader) is invoked in the maps package `CMakeLists` to download the models from [Ignition Fuel](app.ignitionrobotics.org/) hence an internet connection is required for the first build.

codebot, 2022-02-08:
In case you're curious about the details of the Wavefront OBJ and MTL files, the Wikipedia page for that format is a good place to start: https://en.wikipedia.org/wiki/Wavefront_.obj_file

closed 2022-04-20 by codebot