open-rmf/rmf_traffic_editor #420: Map image does not move

ClobotLucy, 2022-02-08:
## Bug report

### Required information:

- Operating system and version:
  - Ubuntu 20.04
- OpenRMF installation type:
  - source
- OpenRMF version or commit hash
  - git main branch
- ROS distribution and version:
  - foxy
- ROS installation type:
  - binaries

### Description of the bug
Map image does not move.
How do I move the image to the origin?

[image]

Changing the value of "flattened x offset(meters)" doesn't move the image.

[image]

Map image does not move to origin even in rviz.
[image]

codebot, 2022-02-08:
Thanks for filing the bug report. So that I can reproduce it easily, can you post the `.building.yaml` file and map image that shows this issue, or PM it to me? My email address is in my GitHub profile: https://github.com/codebot

Cheers! -Morgan

ClobotLucy, 2022-02-08:
Thanks @codebot  
Here is the link with the yaml(untech_office_door.building.yaml) and map(suntech_office.png) files.

[https://github.com/CLOBOT-Co-Ltd/clober_ros2_sim/tree/foxy-devel/maps/suntech_office](https://github.com/CLOBOT-Co-Ltd/clober_ros2_sim/tree/foxy-devel/maps/suntech_office)

codebot, 2022-02-11:
Thanks for linking the repository. This looks like an interesting map!

There are a few reasons for the map behavior you're seeing, and a few ways to move forward.

The "Floorplan" image in this type of map is treated as the coordinate system. In other words, the upper-left pixel (0, 0) of the Floorplan image is treated as the origin of traffic-editor when it's running in `reference_image` mode, which was historically the default mode when creating new maps. The base image, which in this case is https://raw.githubusercontent.com/CLOBOT-Co-Ltd/clober_ros2_sim/foxy-devel/maps/suntech_office/suntech_office.png , cannot be moved, because it is assumed to be the reference frame for the entire map. Often, this floorplan image is a scan or rasterization of the architectural drawings of the building, and then the robot-generated LIDAR maps are placed on top of it. If you have a "known good" floorplan image of this building, I would suggest using that as the Floorplan image, and then positioning these robot-generated maps on top of it.

As an alternative, if you do not have an architectural-drawing floorplan image, and would like to be able to move around several robot-generated maps, you could simply use a blank image (or no image) for the (fixed) floorplan image, and only add robot-generated maps as Layers. Then you could translate/rotate the robot layers as you see fit.

As another alternative, if you build traffic-editor from source and create a new map, it now asks if you'd like to create a "geo-referenced" map when you create a new building. In that case, the fixed "floorplan images" are OpenStreetMap tiles, and all of the images you bring into the map, whether they are architectural drawings, robot maps, or anything else, are placed as layers, and all coordinates in that map are stored as latitude/longitude. There are a few caveats with this brand-new workflow, such as the need to specify a map projection, but it may be more flexible if you don't have a fixed architectural drawing to use as a reference image. Another caveat is that at time of writing, our tile server is only loaded with Singapore and Malaysia map data (I'm not sure which region you're in). We are currently bringing up a new tile server with the whole planet's data, but that will take a few more days.

Cheers!

ClobotLucy, 2022-03-03:
@codebot 
Thank you for your kind explanation.
As you explained, after adding the robot-generated(slam) map as a layer on top of the empty layout, I was able to move the coordinates.
[image]

However, when running the rmf package, there is a problem that the map is not loaded.
May I know what error it is?
[image]

This is new.building.yaml file 
```
coordinate_system: reference_image
crowd_sim:
  agent_groups:
    - {
        agents_name: [],
        agents_number: 0,
        group_id: 0,
        profile_selector: external_agent,
        state_selector: external_static,
        x: 0,
        y: 0,
      }
  agent_profiles:
    - {
        ORCA_tau: 1,
        ORCA_tauObst: 0.4,
        class: 1,
        max_accel: 0,
        max_angle_vel: 0,
        max_neighbors: 10,
        max_speed: 0,
        name: external_agent,
        neighbor_dist: 5,
        obstacle_set: 1,
        pref_speed: 0,
        r: 0.25,
      }
  enable: 0
  goal_sets: []
  model_types: []
  obstacle_set: { class: 1, file_name: floor_navmesh.nav, type: nav_mesh }
  states:
    - { final: 1, goal_set: -1, name: external_static, navmesh_file_name: "" }
  transitions: []
  update_time_step: 0.1
graphs: {}
levels:
  floor:
    elevation: 0
    flattened_x_offset: 0
    flattened_y_offset: 0
    floors:
      - parameters:
          {
            ceiling_scale: [3, 1],
            ceiling_texture: [1, blue_linoleum],
            indoor: [2, 0],
            texture_name: [1, blue_linoleum],
            texture_rotation: [3, 0],
            texture_scale: [3, 1],
          }
        vertices: [4, 1, 2, 3]
    lanes:
      - [
          15,
          16,
          {
            bidirectional: [4, true],
            demo_mock_floor_name: [1, ""],
[... 169 lines cut ...]
            alpha: [3, 1],
            texture_height: [3, 2.5],
            texture_name: [1, default],
            texture_scale: [3, 1],
            texture_width: [3, 1],
          },
        ]
      - [
          13,
          14,
          {
            alpha: [3, 1],
            texture_height: [3, 2.5],
            texture_name: [1, default],
            texture_scale: [3, 1],
            texture_width: [3, 1],
          },
        ]
    x_meters: 10.06924897133468
    y_meters: 10.06924897133468
lifts: {}
name: new
```

Thank you!

Ishansehgal, 2024-10-25:
I am having this same problem i am using ros2 humble

luca-della-vedova, 2024-10-28:
Hi @Ishansehgal,

The `Unable to generate GeoJSON for this map` warning can be safely ignored if you are not creating a geo-referenced map, is there any other issue?