open-rmf/rmf_traffic_editor #439: Failed to install libceres-dev on humble

koonpeng, 2022-07-04:
## Bug report

### Required information:

- Operating system and version:
  - ubuntu 22.04
- OpenRMF installation type:
  - source
- OpenRMF version or commit hash
  - main
- ROS distribution and version:
  - humble
- ROS installation type:
  - binaries
- Package or library, if applicable:
  - libceres-dev

### Description of the bug

The ros ci tooling fail to install `libceres-dev`, logs can be found here https://github.com/open-rmf/rmf/runs/7175224381?check_suite_focus=true#step:5:2526.

While investigating how the ros tooling runs rosdep, I ended up at a script `install_rosdeps.sh` but I wasn't able to find the source of the script. I also wasn't able to reproduce this problem using a docker container, `rosdep keys` and `rosdep resolve` both correctly list `libceres-dev` as one of the install targets.

### Steps to reproduce the bug

Only happens in CI, wasn't able to reproduce locally. Happens consistently in this PR https://github.com/open-rmf/rmf/pull/165.

orensbruli, 2022-08-22:
I'm taking a look into this.
It looks like there's some conflict between the humble dependences installed by the step
```
      - name: deps
        uses: ros-tooling/setup-ros@v0.3
        with:
          required-ros-distributions: humble
```
and the package libceres-dev. 

If libceres-dev is installed in a previous step, it's uninstalled by the ros-tooling/setup-ros@v0.3.
```
| [command]/usr/bin/sudo DEBIAN_FRONTEND=noninteractive RTI_NC_LICENSE_ACCEPTED=yes apt-get install --no-install-recommends --quiet --yes libssl-dev python3-dev build-essential clang cmake git lcov libc++-dev libc++abi-dev python3-catkin-pkg-modules python3-pip python3-vcstool wget libasio-dev libtinyxml2-dev python3-rosdep
| Reading package lists...
| Building dependency tree...
| Reading state information...
| wget is already the newest version (1.21.2-2ubuntu1).
| libssl-dev is already the newest version (3.0.2-0ubuntu1.6).
| git is already the newest version (1:2.37.2-0ppa1~ubuntu22.04.1).
| The following packages were automatically installed and are no longer required:
|   libamd2 libblas-dev libblas3 libbtf1 libcamd2 libccolamd2 libceres2
|   libcholmod3 libcolamd2 libcxsparse3 libeigen3-dev libgflags-dev libgflags2.2
|   libgfortran5 libgoogle-glog0v5 libgraphblas-dev libgraphblas6 libklu1
|   liblapack-dev liblapack3 libldl2 libmetis5 libmongoose2 librbio2 libsliplu1
|   libspqr2 libsuitesparse-dev libsuitesparseconfig5 libumfpack5
| Use 'sudo apt autoremove' to remove them.
| The following additional packages will be installed:
|   clang-14 cmake-data dh-elpa-helper docutils-common emacsen-common
|   lib32gcc-s1 lib32stdc++6 libarchive13 libc++-14-dev libc++1-14
|   libc++abi-14-dev libc++abi1-14 libc6-i386 libclang-common-14-dev
|   libclang-cpp14 libclang1-14 libgc1 libjs-jquery libjs-sphinxdoc
|   libjs-underscore libjson-perl libjsoncpp25 libllvm14 libobjc-11-dev libobjc4
|   libperlio-gzip-perl libpython3-dev libpython3.10 libpython3.10-dev librhash0
|   libtinyxml2-9 libunwind-14 libunwind-14-dev libuv1 llvm-14-linker-tools
|   python3-dateutil python3-docutils python3-roman python3-rosdep-modules
|   python3-rosdistro-modules python3-rospkg-modules python3-setuptools
|   python3-wheel python3-yaml python3.10-dev sgml-base xml-core
| Suggested packages:
|   clang-14-doc cmake-doc ninja-build cmake-format lrzip docutils-doc
|   fonts-linuxlibertine | ttf-linux-libertine texlive-lang-french
|   texlive-latex-base texlive-latex-recommended python-setuptools-doc
|   sgml-base-doc debhelper
| Recommended packages:
|   llvm-14-dev libgd-gd2-perl libboost-regex-dev libboost-dev
|   libboost-date-time-dev javascript-common libjson-xs-perl libpaper-utils
|   python3-pil python3-pygments
| The following packages will be REMOVED:
|   libceres-dev libgoogle-glog-dev libunwind-dev
| The following NEW packages will be installed:
|   build-essential clang clang-14 cmake cmake-data dh-elpa-helper
|   docutils-common emacsen-common lcov lib32gcc-s1 lib32stdc++6 libarchive13
|   libasio-dev libc++-14-dev libc++-dev libc++1-14 libc++abi-14-dev
|   libc++abi-dev libc++abi1-14 libc6-i386 libclang-common-14-dev libclang-cpp14
|   libclang1-14 libgc1 libjs-jquery libjs-sphinxdoc libjs-underscore
|   libjson-perl libjsoncpp25 libllvm14 libobjc-11-dev libobjc4
|   libperlio-gzip-perl libpython3-dev libpython3.10 libpython3.10-dev librhash0
|   libtinyxml2-9 libtinyxml2-dev libunwind-14 libunwind-14-dev libuv1
|   llvm-14-linker-tools python3-catkin-pkg-modules python3-dateutil python3-dev
|   python3-docutils python3-pip python3-roman python3-rosdep
|   python3-rosdep-modules python3-rosdistro-modules python3-rospkg-modules
|   python3-setuptools python3-vcstool python3-wheel python3-yaml python3.10-dev
|   sgml-base xml-core
| 0 upgraded, 60 newly installed, 3 to remove and 8 not upgraded.
```
If installed after, we get the conflict.

orensbruli, 2022-08-23:
Reverse dependencies analysis and conflict path:
In the ros-tooling/setup-ros@v0.3 we are asking to install 
`libc++-dev -> libc++-14-dev -> libc++-14-dev`
but one of the direct dependences of `rmf_traffic_editor` is:
`rmf_traffic_editor->libceres-dev -> libgoogle-glog-dev ->libunwind-dev`

orensbruli, 2022-08-23:
Possible solutions:

- Check if libceres is absolutly needed in the editor, if not check if if can be conditionally imported if exists and set as recommended and not as hard deppendency.
-

luca-della-vedova, 2022-10-05:
From what I understand this should be solved as of https://github.com/ros-tooling/setup-ros/issues/506 ?

geoeo, 2024-07-26:
Still an issue in 2024

Yadunund, 2024-07-26:
> Still an issue in 2024

I don't think this is still an issue. Could you share details of your setup and commands you ran to encounter the error?
 `rosdep` is able to resolve [libceres-dev](https://github.com/ros/rosdistro/blob/70e874f0eba0165e16b5cf9fa732098804d85d20/rosdep/base.yaml#L3110-L3119) in the defined platforms. In fact we wouldn't be able to build binaries of `rmf_traffic_editor` if the dependency cannot be resolved. But you can run `sudo apt install ros-humble-rmf-traffic-editor` to install the binary (assuming you're setup to install ROS packages).

closed 2024-07-26 by Yadunund

geoeo, 2024-07-26:
I installed humble on my system. Now I am trying to install libceres-dev via apt-get and I get the same unmet dependencies message as O.P.

aaronchongth, 2024-07-26:
Sanity check, try running `rosdep update` first

or run `sudo apt install -f`

Please also post your terminal logs when you attempt to do `sudo apt install libceres-dev`

Since we are using the official debian released `libceres-dev`, https://packages.ubuntu.com/jammy/libceres-dev, another suspicion is that you have a pinned version of one of the dependencies of `libceres-dev`, which prevents the simple `apt install` command to update its dependencies, the error logs will provide some clues

geoeo, 2024-07-26:
[image]

I dont think I have anything pinned but ill check

aaronchongth, 2024-07-29:
Unfortunately, there may be something else on your system that is preventing the installation of `libceres-dev`, and not anything related to this issue.

You can verify with a fresh `ros:humble-ros-base-jammy` docker image that `libceres-dev` installs without issue via `apt install`, I'm using `distrobox`

```bash
distrobox create -n test --image ros:humble-ros-base-jammy
distrobox enter test

sudo apt update
sudo apt install libceres-dev
```

[image]

The last resort would be to try on a fresh new installation of ubuntu 22.04

geoeo, 2024-07-29:
ok thanks for your feedback