open-rmf/rmf_traffic_editor #459: How can edit yaw value of points？

mbb1234567890, 2022-12-26:
After the robot reaches a point of interest, it needs to turn the direction angle. How to set these directions in the traffic editor？

[image]

Briancbn, 2022-12-26:
Hi @mbb1234567890, based on my current experience there are two ways to achieve this behavior without modifying the `nav_graph`, it doesn't store any orientation information. The workarounds I did are as follows, depending on the situation.

1. Use a GoToPlace directly (api_request as robot_task) or define a sequence of such and send it to dispatcher instead of Loop task. In the GoToPlace task description, you can define an `orientation` attribute. 

2. Use the dock capability if you still want to use loop task, the robot will reach the point (wp1) leading up the waypoint (wp2) with the `dock_name` defined and call the `dock()` function in the fleet adapter. In this way, before the `dock()` function is called, the robot will be at wp1 and pointing towards wp2.

 but I read from somewhere this feature might get reworked

I would recommend try out the first approach. Maybe the maintainer has better solution and correct me if I am wrong. I am also very interested in how to properly control the end point orientation in rmf.

mbb1234567890, 2022-12-26:
Hi @Briancbn ，I also agree the first approach，but where can define the `orientation` attribute? It can't be hard code for every point. How to make the `nav_graph` to be a semantic map?

Briancbn, 2022-12-27:
> I also agree the first approach，but where can define the `orientation` attribute?

You can try it out by defining it over here like the following
https://github.com/open-rmf/rmf_demos/blob/459919bb8617a8344a7d456c3c054baca6b39b68/rmf_demos_tasks/rmf_demos_tasks/dispatch_go_to_place.py#L101

```python3
        # Define task request description
        go_to_description = {'waypoint': self.args.place,
                             'orientation': 0.0}
```

> How to make the `nav_graph` to be a semantic map?

I don't know the answer to this.

mbb1234567890, 2023-01-09:
Thanks @Briancbn

mbb1234567890, 2023-01-09:
[image]

Yadunund, 2023-01-09:
No, that parameter does not represent a `yaw` value that a robot must respect at that vertex. 

To have you robot arrive at a waypoint with a specific `yaw`, you can add an `orientation` value to the GoToPlace request in your task as @Briancbn has explained above.

If you have any alignment/docking routines you robot can perform to achieve precise positioning/orientation, you may couple the GoToPlace activity with a PerformAction one which will allow you to run any custom routine from your fleet adapter once at that waypoint. Alternatively, you can also assign a `dock` property to that waypoint in `traffic-editor` and have your fleet adapter implement the `dock()` callback to perform any such routine. However, it is recommended not to use the dock method unless the waypoint is a leaf waypoint in the graph, ie, it only connects to one other waypoint and ideally does not overlap with any other traffic lanes.

mbb1234567890, 2023-01-09:
Hi @Yadunund  thanks for explain.
I hope to rely on open-rmf to schedule the robot to head towards the specified direction after arriving at the specified waypoint. I hope that the waypoint and the direction only need to be edited once in the traffic-editor. When the waypoint is a passing point, the robot does not need to consider the direction. When the waypoint is the target point, need to finally turn to the specified direction. 
The GoToPlace solution mentioned above needs to pass in an additional orientation parameter. If this parameter is not edited in xxx.building.yaml, the program cannot automatically obtain the binding relationship.
So I would very much like to edit the direction property of the waypoints in the traffic-editor.

mbb1234567890, 2023-01-11:
Use another configuration file to record waypoint orientation information to solve this problem

closed 2023-01-11 by mbb1234567890