open-rmf/rmf_traffic_editor #461: Advice on setting up an environment with WGS84 coordinates system

alphonsustay, 2023-04-05:
Hi, would like to check what are the requirements needed to setup a traffic-editor environment for using WGS84. For example, getting tiles to display on the editor.

Yadunund, 2023-04-05:
`Building`->`New`->Enter name of file->`Save`->`Geographical Coordinates`->`Save`

If you reopen the editor with the saved file, it should render OSM tiles in the canvas. If you're seeing yellow tiles instead, there might be a problem with our server 🤔

alphonsustay, 2023-04-05:
Hey @Yadunund thanks for the prompt response. I can confirm that I have done the recommended steps and I am seeing the yellow tiles. Can I confirm that the server domain is `https://tiles.sandbox.open-rmf.org`

Setup Environment:
Ubuntu20.04 with ROS2 Galactic. This was tested on branches `galactic-devel` and `main`.

Yadunund, 2023-05-17:
Yeah the server is down so this feature is unfortunately not working at the time. Any fixes will target the new `rmf_site` project and I've opened a ticket there to include this functionality. https://github.com/open-rmf/rmf_site/issues/129