open-rmf/rmf_traffic_editor #484: [Feature request]: Allow multiple thumbnail / assets paths

Briancbn, 2024-02-02:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### Description

It seems that right now the thumbnails only allow 1 path. Allowing more path should help with third-party or creating custom thumbnails that can be maintained in another repo, or for customer sensitive thumbnails.

I think this will be helpful especially I want to upload the demo I did [here](https://github.com/open-rmf/rmf_demos/?tab=readme-ov-file#manufacturing--logistics-world) a long time ago with custom assets to be available instead of upload a lot of extra picture here, that might not be relevant to the use case other people is doing.

### Implementation Considerations

I suggest something like
```
path1:path2
```
which is similar to a `$PATH`
or add in an add button to add in multiple lines of input to add in more path.

### Additional information

If this feature is something that the maintainer also thinks is useful. I would gladly help to make a Pull Request.

Yadunund, 2024-02-03:
I understand the need to support thumbnails for assets not only in `rmf_traffic_editor_assets` but one simple solution is to simlink your custom resource folder to the thumbnails folder that `rmf_traffic_editor` looksup.

> or add in an add button to add in multiple lines of input to add in more path.
In this approach, we also need to support functionality to remove or edit one of multiple paths. At this point we would require a GUI like the `Path Manger` in Windows which allows users to CRUD multiple directories to `$PATH`.
If you're willing to contribute such a feature, we are happy to review the PR.
 
 Ideally, all models are hosted on [Fuel](https://app.gazebosim.org/home) with thumbnails uploaded and we implement some mechanism to preview thumbnails of assets on Fuel. Something like this was already implemented by @luca-della-vedova in our [rmf_site project](https://github.com/open-rmf/rmf_site/pull/149).

luca-della-vedova, 2024-02-05:
Yap to add on to @Yadunund's comment I expect most upcoming feature development to happen on the `rmf_site` codebase, happy to review PRs here but probably won't have bandwidth to contribute major features anytime soon

Briancbn, 2024-02-07:
> Ideally, all models are hosted on [Fuel](https://app.gazebosim.org/home) with thumbnails uploaded and we implement some mechanism to preview thumbnails of assets on Fuel. Something like this was [...]

Good to hear the process on rmf_site! Is there a PR I can reference for this feature?

Yadunund, 2024-02-07:
> Good to hear the process on rmf_site! Is there a PR I can reference for this feature?

Yes, the PR is linked in the original comment.