open-rmf/rmf_traffic_editor #489: [Other issue]: Can't build the rmf workspace

feliperad, 2024-03-08:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### Description

Running ROS humble on Ubuntu 22.04

Starting >>> rmf_demos_dashboard_resources
Starting >>> rmf_demos_assets
Starting >>> rmf_demos_fleet_adapter
Starting >>> rmf_demos_maps                                                  
Starting >>> rmf_demos_tasks
Starting >>> rmf_demos_bridges
Finished <<< rmf_demos_assets [3.16s]                                                                                                                   
Finished <<< rmf_demos_dashboard_resources [3.27s]                                                                                                      
Starting >>> rmf_demos_panel
--- stderr: rmf_demos_maps                                                                                                                                   
Traceback (most recent call last):
  File "/opt/ros/humble/lib/rmf_building_map_tools/building_map_generator", line 33, in <module>
    sys.exit(load_entry_point('rmf-building-map-tools==1.6.2', 'console_scripts', 'building_map_generator')())
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map_generator/building_map_generator.py", line 24, in main
    g.generate_ignition_sdf(
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/generator.py", line 72, in generate_ignition_sdf
    self.generate_sdf(
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/generator.py", line 41, in generate_sdf
    building.generate_sdf_models(output_models_dir)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/building.py", line 537, in generate_sdf_models
    level.generate_sdf_model(model_name, model_path)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/level.py", line 358, in generate_sdf_model
    self.write_sdf(model_name, model_path)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/level.py", line 343, in write_sdf
    self.generate_floors(model_ele, model_name, model_path)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/level.py", line 317, in generate_floors
    floor.generate(
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/floor.py", line 231, in generate
    self.triangulate_polygon(self.polygon, triangles)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/floor.py", line 144, in triangulate_polygon
    for item in geom:
TypeError: 'GeometryCollection' object is not iterable
Traceback (most recent call last):
  File "/opt/ros/humble/lib/rmf_building_map_tools/building_map_generator", line 33, in <module>
    sys.exit(load_entry_point('rmf-building-map-tools==1.6.2', 'console_scripts', 'building_map_generator')())
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map_generator/building_map_generator.py", line 16, in main
    g.generate_gazebo_sdf(
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/generator.py", line 59, in generate_gazebo_sdf
    self.generate_sdf(
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/generator.py", line 41, in generate_sdf
    building.generate_sdf_models(output_models_dir)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/building.py", line 537, in generate_sdf_models
    level.generate_sdf_model(model_name, model_path)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/level.py", line 358, in generate_sdf_model
    self.write_sdf(model_name, model_path)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/level.py", line 343, in write_sdf
    self.generate_floors(model_ele, model_name, model_path)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/level.py", line 317, in generate_floors
    floor.generate(
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/floor.py", line 231, in generate
    self.triangulate_polygon(self.polygon, triangles)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/floor.py", line 144, in triangulate_polygon
    for item in geom:
TypeError: 'GeometryCollection' object is not iterable
Traceback (most recent call last):
  File "/opt/ros/humble/lib/rmf_building_map_tools/building_map_generator", line 33, in <module>
    sys.exit(load_entry_point('rmf-building-map-tools==1.6.2', 'console_scripts', 'building_map_generator')())
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map_generator/building_map_generator.py", line 16, in main
    g.generate_gazebo_sdf(
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/generator.py", line 59, in generate_gazebo_sdf
    self.generate_sdf(
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/generator.py", line 41, in generate_sdf
[... 177 lines cut ...]
    self.write_sdf(model_name, model_path)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/level.py", line 343, in write_sdf
    self.generate_floors(model_ele, model_name, model_path)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/level.py", line 317, in generate_floors
    floor.generate(
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/floor.py", line 231, in generate
    self.triangulate_polygon(self.polygon, triangles)
  File "/opt/ros/humble/lib/python3.10/site-packages/building_map/floor.py", line 144, in triangulate_polygon
    for item in geom:
TypeError: 'GeometryCollection' object is not iterable
gmake[2]: *** [CMakeFiles/generate_office_crowdsim_ign.dir/build.make:77: maps/office_ign/office.world] Erro 1
gmake[1]: *** [CMakeFiles/Makefile2:593: CMakeFiles/generate_office_crowdsim_ign.dir/all] Erro 2
gmake: *** [Makefile:146: all] Erro 2

Failed   <<< rmf_demos_maps [3.50s, exited with code 2]
Aborted  <<< rmf_demos_panel [0.35s]
Aborted  <<< rmf_demos_bridges [3.59s]                                                                                            
Aborted  <<< rmf_demos_fleet_adapter [3.72s]
Aborted  <<< rmf_demos_tasks [3.64s]

Summary: 2 packages finished [4.73s]
  1 package failed: rmf_demos_maps
  4 packages aborted: rmf_demos_bridges rmf_demos_fleet_adapter rmf_demos_panel rmf_demos_tasks
  4 packages had stderr output: rmf_demos_bridges rmf_demos_fleet_adapter rmf_demos_maps rmf_demos_tasks
  3 packages not processed

cardboardcode, 2024-06-01:
> So I was able to successfully build `rmf_demos_maps` with humble binaries. See terminal output below. I believe the reason for build failure above is that you have the pip version of `shapely` installed, ie `2.0` while we strongly recommend using `ubuntu apt` version `1.8` as the dependency is resolved by `rosdep`.
> 
> ```shell
> /opt/ros/humble/lib/python3.10/site-packages/building_map/floor.py:144: ShapelyDeprecationWarning: Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
>   for item in geom:
> ---
> Finished <<< rmf_demos_maps [3min 33s]
> Starting >>> rmf_demos
> Finished <<< rmf_demos [1.28s]                         
> Starting >>> rmf_demos_gz
> Starting >>> rmf_demos_gz_classic                 
> Finished <<< rmf_demos_gz [1.42s]                                                                           
> Finished <<< rmf_demos_gz_classic [1.46s]                         
> 
> Summary: 10 packages finished [3min 40s]
>   5 packages had stderr output: rmf_demos_bridges rmf_demos_fleet_adapter rmf_demos_maps rmf_demos_panel rmf_demos_tasks
> ```
> 
> You can see the deprecation warning from `python3-shapely` but everything should still compile if you use the ubuntu apt version of this python package. Suggest to `python3 -m pip uninstall shapely` and then `sudo apt install python3-shapely` and try again.
> 
> ```shell
> yadunund@ubuntu-22-04:~/ws_rmf_demos$ dpkg -l | grep shapely
> ii  python3-shapely                                    1.8.0-1build1                           amd64        geometric objects, predicates, and operations (Python 3)
> ```
> 
> Having said that, will merge @arjo129's PR into main but I don't think we need to backport it `humble`.

Encountered the same issue trying to build `rmf_demo_maps`, same as thread author. 

Can confirm that reinstalling python package `shapely` as mentioned resolved it for me.

```bash
python3 -m pip uninstall shapely && sudo apt install python3-shapely
```

luca-della-vedova, 2024-11-05:
This should be fixed with the apt package as noted in the comment, but feel free to reopen if it is not the case!

closed 2024-11-05 by luca-della-vedova