open-rmf/rmf_traffic_editor #494: map's location is not correct [Other issue]:

yuduo, 2024-05-29:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### Description

i have a map, from ros, formate is pgm, now i need to change to building.yaml, i find the robot location is not right

[image]

in application, my robot is locate in point7, but now in the website is locate in the left corner

below is my pgm file

[image]

yuduo, 2024-05-29:
[map.zip](https://github.com/open-rmf/rmf_traffic_editor/files/15480428/map.zip)

mxgrey, 2024-05-29:
When you update the position using the RMF API, are you converting the robot coordinates (usually pixel coordinates) to the "canonical" RMF coordinates (which are based on Cartesian meters)?

This is usually done using the [nudged](https://pypi.org/project/nudged/) Python library to align some known pixel coordinates of the robot with known canonical Cartesian coordinates.

yuduo, 2024-05-30:
i use free_fleet_client_ros1 to update the corrdinate

i use below source code 
https://github.com/CLOBOT-Co-Ltd/clober_rmf
it contain server and client

not do any transform, can you give me some examples how to use the nudged?

aaronchongth, 2024-05-30:
Hi @yuduo, you will need to launch the server with the correct transforms filled out here https://github.com/CLOBOT-Co-Ltd/clober_ros2_ff/blob/foxy-devel/launch/clober_ff_server.launch.xml#L26-L29, or any equivalent launch files you are using for the server

to generate these transforms, you can either do it manually by selecting points and passing them to `nudged`, the documentation of the library provides some examples https://pypi.org/project/nudged/. For example, selecting a few points in the RMF map, and selecting those same corresponding points in the robot laser scan, pass them to `nudged` to estimate the transform. We have a small portion of such an example documented https://osrf.github.io/ros2multirobotbook/integration_fleets_adapter_tutorial.html?highlight=nudge#b-configure-transformations-between-rmf-and-robot

Or you can use the `traffic-editor` to overlay your laser scan as a separate layer, create `Feature`s on floor plan layer as well as on the robot laser scan. After attaching `Constraint`s to each set of `Feature`s, the layers should align and you will obtain the transformation in the `.building.yaml` output, like so https://github.com/open-rmf/rmf_demos/blob/main/rmf_demos_maps/maps/office/office.building.yaml#L102-L106. The `office` world in `rmf_demos` is a good example of how that is done

closed 2024-05-31 by yuduo