open-rmf/rmf_traffic_editor #504: [Other issue]: Instructions on how to use align colinear

cardboardcode, 2024-09-03:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### Description

Hi Open-RMF team, may I enquire if more instructions could be provided regarding the use of `align colinear`, `features` and `constraints` as highlighted in #372? 

[image]

[image]

I have searched online, going through discussions, youtube videos as well as the [ROS 2 Multi Robot Book](https://osrf.github.io/ros2multirobotbook/traffic-editor.html). However, don't seem to find any substantial details on it apart from what is covered in the aforementioned Pull Request.

Once again, thank you for maintaining this repository. Based on the instructions, I think I will make a Pull Request to include within ROS 2 Multi Robot Book.

mxgrey, 2024-09-03:
Use click+shift to select a group of vertices and then press the `/` key to tell the traffic editor that you want them to be collinear.

I'm not sure what additional details you'd like. Is there a use case where you're having difficulty applying this, or a situation where it's not working the way you'd expect?

cardboardcode, 2024-09-03:
Apologies. After further debugging, answered my own query.

Think I got confused with my intended use of "aligning" Layers in Traffic Editor using the `features` and `constraints` in order to determine the Transform needed between Grid Cells and RMF.

The intended target feature was `Optimize Layer Transforms` which uses `features` and `constraints`, instead of `align colinear`.

Closing.

Thank you for the quick response @mxgrey

closed 2024-09-03 by cardboardcode

Yadunund, 2024-09-04:
There was a similar ticket in this repo https://github.com/open-rmf/rmf_traffic_editor/issues/502#issuecomment-2284495466. Would you be willing to add documentation to the multirobotbook as suggested to the previous OP?

cardboardcode, 2024-09-04:
Hi @Yadunund

Willing to add to it. Will aim to create the necessary Pull Request sometime next week.