open-rmf/rmf_traffic_editor #505: [Bug]: Lifts and doors not positioned correctly by fiducials

Vortex-TH, 2024-09-03:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### OS and version

Ubuntu 22.04.3

### Open-RMF installation type

Binaries

### Open-RMF version or commit hash

https://github.com/open-rmf/rmf/releases/tag/release-iron-231229

### ROS distribution

Iron

### ROS installation type

Binaries

### Description of the bug

I had some relocalization and map issues when my bot arrives at L2 (coming from L1 with the lift), so I did some tests with the hotel demo, which makes me think there are some bugs with the map building tools.

I am using the iron branch of rmf_demos and started the hotel demo, which works fine:
ros2 launch rmf_demos_gz hotel.launch.xml

I then did the following:

### Steps to reproduce the bug

1. Changed the drawing image for L2 by rotating it 45° (Image 1).
2. In traffic-editor, load L2, then adjust all objects, nodes, fiducials, etc. according to the drawing (Image 2).
3. Reload traffic-editor, expecting that the position (and orientation, scale) of the lifts is automatically adjusted according to the changed fiducials
4. Notice that only the position of the lift was adjusted, but not orientation (and probabably scale, which I didn't check) (Image 3)
5. After building and launching the hotel demo again, I notice that lifts (and apparently doors) didn't move in RViz. The 3D model in Gazebo looks fine though. (Image 4 and 5)

### Expected behavior

The lift and doors should be correctly positioned.

### Additional information or screenshots

<details>
  <summary>Image 1</summary>

[image]
</details>
<details>
  <summary>Image 2</summary>

[image]
</details>
<details>
  <summary>Image 3</summary>

[image]
</details>
<details>
  <summary>Image 4</summary>

[image]
</details>
<details>
  <summary>Image 5</summary>

[image]
</details>

aaronchongth, 2024-09-04:
Hello @Vortex-TH! Just to make sure I understand the problem clearly, the issue here is currently related to the rviz visualizations, where the doors and lift positions are not aligned with the floor plan?

Could you share the modified `.building.yaml` for the rotated hotel world, so we can look into it locally?

Yeah the simulation looks alright, have you tried running some patrol tasks with the robots, such that they attempt to go through doors and lifts? That can help narrow down where the issue is.

rmf visualizations on rviz is generally using the same building map file that generates the simulation, so there should not be any discrepancies. Some sanity checks,
* run some patrol tasks that go through door and lifts, and verify if the simulation works, this validates that the navigation graph generated from the building map works properly with the building map
* just in case this is caused by some artifact in old DDS messages, can you try running with cyclonedds? https://docs.ros.org/en/humble/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html

Vortex-TH, 2024-09-04:
Sure, here is my updated hotel.building.yaml and also my modified hotel_L2.png

<details>
<summary>hotel.building.yaml</summary>

```
coordinate_system: reference_image
crowd_sim:
  agent_groups:
    - {agents_name: [cleanerBotA_2, cleanerBotA_1, tinyBot_1, deliveryBot_1], agents_number: 4, group_id: 0, profile_selector: external_agent, state_selector: external_static, x: 0, y: 0}
  agent_profiles:
    - {ORCA_tau: 1, ORCA_tauObst: 0.40000000000000002, class: 1, max_accel: 0, max_angle_vel: 0, max_neighbors: 10, max_speed: 0, name: external_agent, neighbor_dist: 5, obstacle_set: 1, pref_speed: 0, r: 0.25}
  enable: 0
  goal_sets: []
  model_types: []
  obstacle_set: {class: 1, file_name: L1_navmesh.nav, type: nav_mesh}
  states:
    - {final: 1, goal_set: -1, name: external_static, navmesh_file_name: ""}
  transitions: []
  update_time_step: 0.10000000000000001
graphs:
  {}
levels:
  L1:
    doors:
      - [40, 39, {motion_axis: [1, start], motion_degrees: [3, 90], motion_direction: [2, -1], name: [1, L1_door1], plugin: [1, normal], right_left_ratio: [3, 1], type: [1, double_hinged]}]
      - [10, 9, {motion_axis: [1, start], motion_degrees: [3, 90], motion_direction: [2, 1], name: [1, L1_door2], plugin: [1, normal], right_left_ratio: [3, 1], type: [1, double_sliding]}]
      - [35, 34, {motion_axis: [1, start], motion_degrees: [3, 90], motion_direction: [2, 1], name: [1, L1_door3], plugin: [1, normal], right_left_ratio: [3, 1], type: [1, double_sliding]}]
      - [49, 52, {motion_axis: [1, start], motion_degrees: [3, 90], motion_direction: [2, -1], name: [1, L1_door4], plugin: [1, normal], right_left_ratio: [3, 1], type: [1, hinged]}]
      - [47, 45, {motion_axis: [1, start], motion_degrees: [3, -90], motion_direction: [2, 1], name: [1, L1_door5], plugin: [1, normal], right_left_ratio: [3, 1], type: [1, double_sliding]}]
    drawing:
      filename: hotel_L1.png
    elevation: 0
    fiducials:
      - [275.62799999999999, 469.55099999999999, ref3]
      - [274.93400000000003, 304.34699999999998, ref1]
      - [383.59300000000002, 367.916, ref2]
    floors:
      - parameters: {ceiling_scale: [3, 1], ceiling_texture: [1, blue_linoleum], indoor: [2, 0], texture_name: [1, concrete2], texture_rotation: [3, 0], texture_scale: [3, 1]}
        vertices: [44, 43, 36, 37, 38, 39, 40, 41, 42, 0]
      - parameters: {ceiling_scale: [3, 1], ceiling_texture: [1, blue_linoleum], indoor: [2, 0], texture_name: [1, red_tiles], texture_rotation: [3, 0], texture_scale: [3, 1]}
        vertices: [1, 2, 3, 4, 5, 6, 7, 8]
      - parameters: {ceiling_scale: [3, 1], ceiling_texture: [1, blue_linoleum], indoor: [2, 0], texture_name: [1, white_beige_tiles], texture_rotation: [3, 0], texture_scale: [3, 1]}
        vertices: [33, 32, 28, 27, 26, 25, 24, 6, 7, 8, 1, 0, 42, 41, 40, 39, 38, 37, 36]
    lanes:
      - [56, 57, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 0], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [56, 58, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 0], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [58, 59, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 0], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [57, 60, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 0], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [61, 62, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 0], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [61, 55, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 0], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [67, 68, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 1], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [71, 72, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 1], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [72, 76, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 1], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [76, 77, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 1], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [77, 78, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 1], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [62, 82, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 0], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [83, 68, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 1], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
      - [83, 69, {bidirectional: [4, true], demo_mock_floor_name: [1, ""], demo_mock_lift_name: [1, ""], graph_idx: [2, 1], mutex: [1, ""], orientation: [1, ""], speed_limit: [3, 0]}]
[... cut ...]
<summary>hotel_L2.png</summary>

[image]
</details>

The first issue is that the lift position/rotation is obviously not updated correctly in traffic editor for L2.

And yes, the positions in Rviz are not displayed correctly (maybe this is a seperate issue).

I didn't do any robot tasks because I would assume that the robot positions in L2 are incorrect anyway because in the demo there are no level specific reference coordinates in the robot config for the coordinate transformations. But I'll see if I can get it to work.

aaronchongth, 2024-09-04:
I see what you mean now

> The first issue is that the lift position/rotation is obviously not updated correctly in traffic editor for L2.

The lift position is not tied to the waypoints or vertices in the traffic editor, hence it will not be updated. The only connection is the designated waypoint within the lift. You will just need to adjust the lift position with the lift tab, for example

[image]

Vortex-TH, 2024-09-04:
But then it's wrong in L1 and L3, that's the whole point. Or am I missing something here?

[image]

Vortex-TH, 2024-09-05:
@aaronchongth Could you have another look at this? My assumption was that I only position the lift once in L1, because that is the reference floor. The position, orientation and scale of the lift on the other floors should then be automatically derived from that by the traffic editor based on the fiducial markers.

aaronchongth, 2024-09-06:
I see what you mean about the orientation across levels, I'll reopen this issue and take a look at it

TanJunKiat, 2024-10-09:
HI @aaronchongth, I opened up a PR with the graphic fix for the lift cabin.

We still need to fix the alignment for the `lift_cabin` vertex but I want to discuss on the approach to take on this.

Should we implement the drawing of `lift_cabin` vertices within the `vertex.draw` function or should we implement it in the `lift.draw` function itself.

The consideration is that for the first method, we will need to pass the `transform` variable from all the lifts' reference level to the current viewport's level downstream to the `vertex.draw` function which can be messy.

The second method might be cleaner but might affect the interaction with the vertex, such as being unable to click and select the `lift_cabin` vertex.

Please advice.

aaronchongth, 2024-10-10:
Hey @TanJunKiat! Thanks for adding the fix, I'll be looking into it momentarily.

Off the top of my head, I would agree that the second method would be ideal, and semantically more accurate too. As for being able to select the lift vertex on the UI or not, if it comes to it, the backup plan could be to set the `lift_cabin` vertex as `selected` when the lift is selected on the right-top `lifts` panel.

I'll go through the changes and put up any other comments I have in the PR

NgNhatHai, 2024-10-24:
@aaronchongth hi! Have you updated this issue? I also have this issue and have no solution yet.
I scan map floor to get L1.png
[image]
And  map L2.png 
[image]
Two map are the same floor 

and You can see position and rotation of Lift is wrong
Please advice.

TanJunKiat, 2024-11-06:
Hi @aaronchongth 

Some updates on this.

After thinking about it and looking at the code, I feel there is a better approach to the problem on lift cabin alignment.

The baseline is that vertices are defined in the level's coordinate in the `building.yaml` file. This means that we cannot be drawing the vertices in the `lift.draw()` because the elements of the lift are drawn in the lift's local frame before shifting it to the level's coordinate.

The objective for the lift cabin alignment is to get the coordinates of the lift cabin vertex at the reference level of each lift, and apply the level transform across levels.

We also need to ensure that the users can only move the lift cabin vertex at the reference level and allow the fiducials to handle the other levels.

I uploaded a commit based on these and the following pointers are the summary of change:
- Modified the `move' button logic to check if selected vertex is a lift cabin
   - If yes, check if vertex is at reference level
     - If no, return prematurely to prevent moving of vertex
 - Modified the `move'  logic to align lift cabin vertex
   - If a lift cabin vertex of a lift is moved, modify other level's vertex of said lift
- Create a function in the `Building` class to return a `Lift` class for a given lift name.

I will see what else can be optimize in the new codes and check for potential bugs.

mxgrey, 2024-11-15:
This issue is now fixed by #513

This one was a community effort, so I want to give special thanks to @Vortex-TH for the detailed and reproducible report, @aaronchongth for triaging the problem, and @TanJunKiat for delivering the solution. Community contributions like this are greatly appreciated!