open-rmf/rmf_traffic_editor #506: [Bug]: Lift Name stored as number instead of string in .building.yaml

cardboardcode, 2024-09-06:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### OS and version

Ubuntu 22.04

### Open-RMF installation type

Binaries

### Open-RMF version or commit hash

 release-humble-231229

### ROS distribution

Humble

### ROS installation type

Binaries

### Description of the bug

## **Issue Description** :bug: 
When creating a lift and setting `lift name`, naming the lift a number causes runtime error when feeding the generated `.building.yaml` file from RMF Traffic Editor to `building_map_server`.

[image]

The error is reported as such:

```bash
[building_map_server-13] Traceback (most recent call last):
[building_map_server-13]   File "/opt/ros/humble/lib/rmf_building_map_tools/building_map_server", line 33, in <module>
[building_map_server-13]     sys.exit(load_entry_point('rmf-building-map-tools==1.6.2', 'console_scripts', 'building_map_server')())
[building_map_server-13]   File "/opt/ros/humble/lib/python3.10/site-packages/building_map_server/building_map_server.py", line 339, in main
[building_map_server-13]     n = BuildingMapServer(map_path)
[building_map_server-13]   File "/opt/ros/humble/lib/python3.10/site-packages/building_map_server/building_map_server.py", line 49, in __init__
[building_map_server-13]     self.load_building_yaml(map_path)
[building_map_server-13]   File "/opt/ros/humble/lib/python3.10/site-packages/building_map_server/building_map_server.py", line 87, in load_building_yaml
[building_map_server-13]     self.create_map_msg(building)
[building_map_server-13]   File "/opt/ros/humble/lib/python3.10/site-packages/building_map_server/building_map_server.py", line 107, in create_map_msg
[building_map_server-13]     self.map_msg.lifts.append(self.lift_msg(lift_data))
[building_map_server-13]   File "/opt/ros/humble/lib/python3.10/site-packages/building_map_server/building_map_server.py", line 286, in lift_msg
[building_map_server-13]     msg.name = lift.name
[building_map_server-13]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rmf_building_map_msgs/msg/_lift.py", line 180, in name
[building_map_server-13]     assert \
[building_map_server-13] AssertionError: The 'name' field must be of type 'str'
```

### Steps to reproduce the bug

1. Open RMF Traffic Editor
2. Create a map with a lift and name it `1`.
3. Generate the `.building.yaml` RMF Map file.
4. Feed the `.building.yaml` RMF Map file to `building_map_server`

### Expected behavior

No runtime error should be encountered. The map loads successfully and is displayed on an RViz window.

### Actual behavior

The above runtime error is encountered. Map fails to display in RViz window.

Yadunund, 2024-09-06:
This is a known issue https://github.com/open-rmf/rmf_traffic_editor/issues/462.

Pull request with a fix is welcome!

cardboardcode, 2024-09-07:
Noted. Will aim to make a PR with a fix once my work bandwidth allows me. 

Will update here once I have made PR. 

Closing issue for now.

closed 2024-09-07 by cardboardcode

cardboardcode, 2024-09-07:
## **Update** :speech_balloon: 

Implemented fix in #507