open-rmf/rmf_traffic_editor #512: Not found models in Fuel ignitionrobotics

harunkurtdev, 2024-09-28:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### Description

```bash
ros2 run rmf_building_map_tools building_map_model_downloader ../workshop_example/workshop_example.building.yaml -e .
building name: workshop
coordinate system: reference_image
parsing level L1
parsing level L2
calculating levels relative to L1
level L1 scale: 0.025388768725657793
calculating level L2 offset and scale...
  2 common fiducials:
Transform.set_from_fiducials()
  2 fiducial pairs:
    (487.54, -1418.1) -> (487.54, -1418.1)   left_column
    (1119.7, -1418.1) -> (1119.7, -1418.1)   right_column
    calc dist left_column <=> right_column
Bearings:
[(0.0, 0.0)]
  0.0
  Circular mean: -0.0
Distances:
[(632.1440000000001, 632.1440000000001)]
mean relative scale: 1.0
translation: (-0.0, -0.0)
parsing lift lift
Models: [] are used in crowd_sim
Load success!

Cache found! Model count: 137 
Updating cache...
Topping up Fuel (updating cache) from: https://fuel.ignitionrobotics.org/1.0/models
Fetching page: 1
Cached model found! Halting Fuel traversal...
Writing to cache: /home/harunkurtdev/.pit_crew/model_cache.json
New models cached: 0

== REQUESTED MODEL REPORT ==
{'available': [],
 'downloadable': [],
 'missing': ['Sofa', 'AdjTable', 'OfficeChairGrey', 'LitterBin']}

== THE FOLLOWING MODELS HAVE SPECIFIED AUTHORS ==
{'adjtable': 'openrobotics',
 'litterbin': 'openrobotics',
 'officechairgrey': 'openrobotics',
 'sofa': 'openrobotics'}

WARNING::building_map_model_downloader.download_models():164: 
Missing models (not in local or Fuel):
['Sofa', 'AdjTable', 'OfficeChairGrey', 'LitterBin']
```

aaronchongth, 2024-09-30:
Hi there, perhaps there were issues with the fuel server during that period of time, I've just managed to attempt what you did without issues

Just to make sure I don't use any cached models, I cleared `~/.gazebo/models` and `~/.pit_crew` ahead of time

```bash
rm -rf ~/.gazebo/models ~/.pit_crew

git clone https://github.com/open-rmf/roscon_workshop

mkdir -p roscon_workshop/roscon_maps/maps/test
cd roscon_workshop/roscon_maps/maps/test

source MY_RMF_WORKSPACE/install/setup.bash

ros2 run rmf_building_map_tools building_map_model_downloader ../workshop_example/workshop_example.building.yaml -e .
```

This was the output

```
building name: workshop                                                                                                                     
coordinate system: reference_image                                                                                                          
parsing level L1                                                                                                                            
parsing level L2                                                                                                                            
calculating levels relative to L1                                                                                                           
level L1 scale: 0.025388768725657793                                                                                                        
calculating level L2 offset and scale...                                                                                                    
  2 common fiducials:                                                                                                                       
Transform.set_from_fiducials()                                                                                                              
  2 fiducial pairs:                                                                                                                         
    (487.54, -1418.1) -> (487.54, -1418.1)   left_column                                                                                    
    (1119.7, -1418.1) -> (1119.7, -1418.1)   right_column                                                                                   
    calc dist left_column <=> right_column                                                                                                  
Bearings:                                                                                                                                   
[(0.0, 0.0)]                                                                                                                                
  0.0                                                                                                                                       
  Circular mean: -0.0                                                                                                                       
Distances:                                                                                                                                  
[(632.1440000000001, 632.1440000000001)]                                                                                                    
mean relative scale: 1.0                                                                                                                    
translation: (-0.0, -0.0)                                                                                                                   
parsing lift lift                                                                                                                           
Models: [] are used in crowd_sim                                                                                                            
Cache directory does not exist! Creating it: /usr/local/google/home/aaronchong/.pit_crew                                                    
Cache not found! Rebuilding cache...                                                                                                        
Topping up Fuel (updating cache) from: https://fuel.gazebosim.org/1.0/models                                                                
Fetching page: 1                                                                                                                            
Fetching page: 2                                                                                                                            
Fetching page: 3                                                                                                                            
Fetching page: 4                                                                                                                            
Fetching page: 5                                                                                                                            
Fetching page: 6                                                                                                                            
Fetching page: 7                                                                                                                            
Fetching page: 8                                                                                                                            
[... cut ...]
== THE FOLLOWING MODELS HAVE SPECIFIED AUTHORS ==
{'adjtable': 'openrobotics',
 'litterbin': 'openrobotics',
 'officechairgrey': 'openrobotics', 
 'sofa': 'openrobotics'}

Exporting to ./OfficeChairGrey...
Exporting to ./Sofa...
Exporting to ./LitterBin...
Exporting to ./AdjTable...
```

I'd suggest trying again, with your current setup, and if it fails again, attempt it again after backing up and clearing your caches

```bash
mv ~/.gazebo/models ~/.gazebo/models-bak
mv ~/.pit_crew ~/.pit_crew-bak
```

closed 2024-09-30 by aaronchongth

harunkurtdev, 2024-10-04:
your comment showing  url
```bash
https://fuel.gazebosim.org/1.0/models 
```
my url like this 
```bash
https://fuel.ignitionrobotics.org/1.0/models
```

I download reagain rmf_traffic_editor repo, I did.

Thank you.

aaronchongth, 2024-10-07:
Looks like you're still using an older version of gazebo, you can verify by running `gz sim --versions` to check which versions you have installed. If you're running Ubuntu 24.04, and ROS 2 Jazzy, you'd want to have Gazebo Harmonic, which is version 8.6.0 as of now.

 you can try uninstalling it and installing gazebo `harmonic` manually, https://gazebosim.org/docs/harmonic/install_ubuntu/