open-rmf/rmf_traffic_editor #516: [Other issue]: How can i decrease lane width in traffic editor...

pvg6, 2024-11-22:
### Before proceeding, is there an existing issue or discussion for this?

- [X] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### Description

I want to do openemf in very small place which is 2X3 meters... when i create the map in the traffic editor. the wapoints only occupy the map..

[image]

please help me with this...

TanJunKiat, 2024-11-22:
Hello, this is a visualization preference set in 'rmf_visualization', under the 'rmf_visualization_navgraphs' package. The marker sizes are hardcoded to a specific value at the moment. 

You can tweak the waypoint and lane marker size to a smaller value to make it more readable when running.

pvg6, 2024-11-22:
@TanJunKiat  Thank you.. solved it..

closed 2024-11-22 by pvg6