open-rmf/rmf_traffic_editor #529: [Bug]: Missing models when building `rmf_demos_maps`

TanJunKiat, 2025-04-24:
### Before proceeding, is there an existing issue or discussion for this?

- [x] I have done a search for similar [issues](.?q=type%3Aissue) and [discussions](https://github.com/open-rmf/rmf/discussions).

### OS and version

Ubuntu 24.04

### Open-RMF installation type

Source build

### Open-RMF version or commit hash

main

### ROS distribution

Jazzy

### ROS installation type

Source build

### Description of the bug

Reference to https://github.com/open-rmf/rmf_demos/pull/291, some users are experiencing issues when building the `rmf_demos_maps`, where models are not getting downloaded from online.

Here is a table of what I have with the previous build of OpenRMF vs what is downloaded now:

|Model|Status|
|-|-|
|AdjTable|:white_check_mark:|
| Ambulance2 |:x:|
| BathroomSink |:white_check_mark:|
| BedsideTable |:white_check_mark:|
| BigCubicle |:x:|
| Bookshelf |:white_check_mark:|
| Cafe |:x:|
| Cafe table|:white_check_mark:|
| Campus |:x:|
| CGMVanguard|:white_check_mark:|
| Chair |:x:|
| CoffeeTable |:white_check_mark:|
| Coke|:white_check_mark:|
| ConfTable |:x:|
| Construction Cone |:white_check_mark:|
| Desk |:white_check_mark:|
| DeskChair|:white_check_mark:|
| DigitalKiosk |:white_check_mark:|
| Drawer|:white_check_mark:|
| ElderLadyPatient |:white_check_mark:|
| ElderMalePatient |:x:|
| ElectronicRecycling |:white_check_mark:|
| EscalatorStart |:x:|
| FemaleVisitor|:white_check_mark:|
| FemaleVisitorSit|:white_check_mark:|
| FoodCourtBenchShort |:white_check_mark:|
| FoodCourtTable1|:white_check_mark:|
| Fountain |:x:|
| Fridge|:white_check_mark:|
| Gazebo|:white_check_mark:|
| HeadphoneRack1|:white_check_mark:|
| IVStand |:white_check_mark:|
| KitchenCountertop |:x:|
| KitchenSink |:white_check_mark:|
| LampAndStand |:white_check_mark:|
| Lamp Post |:x:|
| LitterBin |:white_check_mark:|
| MainTable |:x:|
| MaleVisitorOnPhone |:x:|
| MaleVisitorPhone |:x:|
| MaleVisitorSit |:x:|
| MetalCabinet  |:white_check_mark:|
| MetalCabinetYellow |:white_check_mark:|
| MiniSofa|:white_check_mark:|
| MiR100 |:x:|
| MonitorAndKeyboard |:white_check_mark:|
| MopCart2 |:white_check_mark:|
| MopCart3 |:x:|
| NormalBed|:white_check_mark:|
| Nurse |:white_check_mark:|
| NurseDesk |:white_check_mark:|
| NurseTable |:white_check_mark:|
| Oak tree |:white_check_mark:|
| OfficeChairBlack |:x:|
| OfficeChairBlue |:white_check_mark:|
| OfficeChairGrey |:white_check_mark:|
| OpScrubs |:white_check_mark:|
| ParkingTrolleyMax |:x:|
|  PatientFSit |:white_check_mark:|
| PatientWheelChair |:white_check_mark:|
| Pine Tree |:x:|
| Playground |:white_check_mark:|
| PotatoChipChair |:white_check_mark:|
| RecTable |:x:|
| Scrubs |:x:|
| Shower |:x:|
| SmallCubicle |:x:|
| Sofa |:x:|
| SquareShelf |:white_check_mark:|
| StorageRack |:white_check_mark:|
| Suitcase 1 |:x:|
| Suitcase2 |:white_check_mark:|
| SurgicalTrolleyMed |:white_check_mark:|
| SUV |:white_check_mark:|
| Table |:white_check_mark:|
| Toilet |:white_check_mark:|
| TrashBin |:white_check_mark:|
| VendingMachine |:white_check_mark:|
| VisitorKidSit |:white_check_mark:|
| WhiteCabinet|:white_check_mark:|
| WhiteChipChair |:white_check_mark:|
| WoodenChair |:white_check_mark:|

### Steps to reproduce the bug

1. Delete existing models in `$HOME/.gazebo/models`
2. Clean build `rmf_demos_maps`
3. Try to launch any of the simulations (Office was used for me)
4. Gazebo crashes with missing model error 
```terminal
[Err] [Server.cc:86] Error Code 14: [/sdf/world[@name="sim_world"]/include[61]/uri:/home/parallels/rmf_ws/install/rmf_demos_maps/share/rmf_demos_maps/maps/office/office.world:L701]: Msg: Unable to find uri[model://RecTable]
```

### Expected behavior

Simulations to be launched and gazebo running normally.

### Actual behavior

Gazebo crashes.

mxgrey, 2025-04-24:
I can confirm that I'm able to reproduce this issue, although interestingly I'm getting a different set of models that are missing vs available. I wonder if this is actually an issue with availability of the gazebo fuel server?

mxgrey, 2025-04-24:
I'm attempting to recreate this again on a different machine with a totally clean build of Open-RMF.

This time the issue with downloading the models is presenting as build errors, e.g.:

```
ERROR::pit_crew.download_model():596: Could not download model 'FemaleVisitor'! ('Connection broken: IncompleteRead(462369 bytes read, 223147 more expected)', IncompleteRead(462369 bytes read, 223147 more expected))
WARNING::building_map_model_downloader.download_models():183: Retrying 0 of 5 times...
WARNING::pit_crew.download_model():531: No path given! Downloading to ~/.gazebo/models instead!
ERROR::pit_crew.download_model():596: Could not download model 'FemaleVisitor'! ('Connection broken: IncompleteRead(257610 bytes read, 427906 more expected)', IncompleteRead(257610 bytes read, 427906 more expected))
WARNING::building_map_model_downloader.download_models():183: Retrying 1 of 5 times...
WARNING::pit_crew.download_model():531: No path given! Downloading to ~/.gazebo/models instead!
ERROR::pit_crew.download_model():596: Could not download model 'FemaleVisitor'! ('Connection broken: IncompleteRead(319226 bytes read, 366290 more expected)', IncompleteRead(319226 bytes read, 366290 more expected))
WARNING::building_map_model_downloader.download_models():183: Retrying 2 of 5 times...
WARNING::pit_crew.download_model():531: No path given! Downloading to ~/.gazebo/models instead!
ERROR::pit_crew.download_model():596: Could not download model 'FemaleVisitor'! ('Connection broken: IncompleteRead(403065 bytes read, 282451 more expected)', IncompleteRead(403065 bytes read, 282451 more expected))
WARNING::building_map_model_downloader.download_models():183: Retrying 3 of 5 times...
WARNING::pit_crew.download_model():531: No path given! Downloading to ~/.gazebo/models instead!
ERROR::pit_crew.download_model():596: Could not download model 'FemaleVisitor'! ('Connection broken: IncompleteRead(449635 bytes read, 235881 more expected)', IncompleteRead(449635 bytes read, 235881 more expected))
WARNING::building_map_model_downloader.download_models():183: Retrying 4 of 5 times...
ERROR::building_map_model_downloader.download_models():186: Failed to download model [%s/%s] after 5 attempts, exiting...
``` 

I think this confirms my suspicion that the problem is related to the fuel server. I find it interesting that we're getting incomplete reads, which implies that some of the data is transferred but it gets cut off partway through..?

mxgrey, 2025-04-24:
I'm hearing from the core Gazebo team that the Gazebo fuel server is recently experiencing some connection issues more broadly than just for Open-RMF.

For now let's hang tight until they can investigate.

One option that might be worth considering is to have a separate git repo dedicated to the assets that we normally fetch from fuel. We wouldn't include that repo in the `rmf.repos` file, but at least it would be available as a backup in case outages like this occur again in the future.

shamlymackey, 2025-04-24:
@mxgrey same for me. I got a completely different set of models in 2 different times I tried to build. But both times I got less than 10 models in total. For any one running into this issue, a temporary fix is to download all the necessary models manually. I forked a rmf_demos and have added all the models necessary. Hopefully this helps anyone to get going: https://github.com/shamlymackey/rmf_demos

TanJunKiat, 2025-06-02:
Closing issue. Seems to be working now.

closed 2025-06-02 by TanJunKiat