open-rmf/rmf_traffic_editor #539: UserWarning: Unknown distribution option: 'tests_require'

Plumezz, 2025-07-14:
- ROS Version: Humble

- OS Version: Ubuntu22.04

- Python Version: 3.10.12

- setuptools Version: 75.2.0

Hello, when I build this project from source, some warnings occurred.

Hers is the output of the command `colcon build`:
```
Starting >>> rmf_building_map_tools
Starting >>> rmf_traffic_editor
Starting >>> rmf_traffic_editor_assets
/home/qiguanxiao/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/home/qiguanxiao/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
--- stderr: rmf_building_map_tools                                   
/home/qiguanxiao/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
---
Finished <<< rmf_building_map_tools [3.52s]
Starting >>> rmf_traffic_editor_test_maps
--- stderr: rmf_traffic_editor_assets                                     
/home/qiguanxiao/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
---
Finished <<< rmf_traffic_editor_assets [3.95s]
```

This warning shows the `tests_require` option is expired. But the python and setuptools are default package with the ubuntu 22.04.

And then I use `colcon test` to test this project, there were also some warnings.

Here is the output about the command `colcon test`:
```
Starting >>> rmf_building_map_tools
Starting >>> rmf_traffic_editor
Starting >>> rmf_traffic_editor_assets
/home/qiguanxiao/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
/home/qiguanxiao/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
--- stderr: rmf_traffic_editor_assets                                

----------------------------------------------------------------------
Ran 0 tests in 0.000s

OK
---
Finished <<< rmf_traffic_editor_assets [0.62s]
--- stderr: rmf_building_map_tools

----------------------------------------------------------------------
Ran 0 tests in 0.000s

OK
---
Finished <<< rmf_building_map_tools [0.68s]
```

It seems like that there were no python testing programs to run and no result file `pytest.xml` was generated in build directory.

Then I tried to modify the `setup.py` file in package `rmf_building_map_tools, rmf_traffic_editor_assets` and replaced the option `tests_require` with `extras_require`.

Finally it can be built and test correctly.

Here is the modified content in `setpu.py` file for package `rmf_building_map_tools`:
```
tests_require=['pytest'],  # remove
extras_require={
        'test': ['pytest'],
 },  # add
```

For package `rmf_traffic_editor_assets`, I noticed that there were no tests so I removed this line in its `setup.py`:
```
tests_require=['pytest'],  # remove
```

Now the python and setuptools version are updated very quickly, could you tell me if it is necessary to replace the `tests_require` option to adapt to the current environment.

Thanks!

Plumezz, 2025-07-14:
Hi, thanks for your answer.

I know this issue is because the version of my `setuptools` is too high. 

https://robotics.stackexchange.com/questions/101255/setuptoolsdeprecationwarning-setup-py-install-is-deprecated-use-build-and-pip  also mentions one solution : `python` version 3.8, `setuptools` version 58.2.0.

But now `python 3.10.12` and `setuptools 75.2.0` that I am using is the default package of Ubuntu 22.04. So I replaced the `tests_require` option without installing a lower version tool.

Now the `python` and `setuptools` version are updated very quickly, could you tell me if it is necessary to replace the `tests_require` option to adapt to the current environment instead of installing a lower version tool which is not the default package in Ubuntu 22.04 or a newer os version.

Thanks!

aaronchongth, 2025-07-14:
Can you check your installation of `setuptools` with the commands in the example below?

```
root@5dbacd2c4e21:/# python3
Python 3.10.12 (main, May 27 2025, 17:12:29) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import setuptools
>>> setuptools.__version__
'59.6.0'
>>> setuptools.__file__
'/usr/lib/python3/dist-packages/setuptools/__init__.py'
```

Are you perhaps using a python virtual environment? By default, the binaries of `python3-setuptools` for Ubuntu 22.04 Jammy is only released up to 59.6.0, https://launchpad.net/ubuntu/jammy/+package/python3-setuptools

Using a python virtual environment or mixing pip packages with ubuntu installed binaries is discouraged, these binaries are built against other released binaries, and therefore may cause undefined behavior when sources are mixed

Plumezz, 2025-07-15:
Thanks for your detailed explainationThanks for your detailed explanation.

Here is my `setuptools`:

```
qiguanxiao@qiUbuntu:~/Desktop$ python3
Python 3.10.12 (main, May 27 2025, 17:12:29) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import setuptools
>>> setuptools.__version__
'75.2.0'
>>>
```

I think I know where the problem lies. I used apt to install python and pip to install setuptools with command `pip3 install setuptools`. So I thought their versions were corresponding.

I am deeply sorry for my careless mistake.

But I notice that even the `setuptools 59.6.0` was relased at Dec 13, 2021, which is now almost four years ago. Moreover, in the newer versions, the `tests_require` option has been deprecated. Would you have any thoughts or suggestions on how to address or improve this related issue?

Thanks!

aaronchongth, 2025-07-15:
Good to know that we've narrowed down the problem!

The older release used is generally due to the age of the distro, as packages tend to try to be backward compatible within each distro, so any breaking changes are only introduced in future releases. For example in Ubuntu 24.04, https://launchpad.net/ubuntu/noble/+source/setuptools, the released version is 68.1.2. The way to go would be to use a newer distribution in general, you will get access to newer features, enhancements and bug fixes too.